OpenTabletDriver/OpenTabletDriver #58: XP-Pen G640 cannot be opened

nerd, 2019-12-15:
## Description
Trying to use an XP-Pen G640 tablet results in a DeviceIOException stating that the device cannot be opened.

## System Information:
Option            | Value
------------------|------------------
Operating System  |  Windows 10
Software Version  | 0.1.0-1301bbb
Tablet            |  XP-Pen G640

InfinityGhost, 2019-12-18:
Installing the XP-Pen drivers seems to unlock the HID stream, allowing it to be opened. Hopefully I can make this not a requirement, but it may come down it that its required.

InfinityGhost, 2019-12-21:
Issue is noted in the [wiki](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Windows-FAQ#my-xp-pen-tablet-fails-to-open-deviceioexception-unable-to-open-hid-class-device) until a further fix is found.

selplacei, 2020-01-04:
Experiencing the same bug on Arch Linux, kernel version 5.4.7-arch1-1

MoeLava, 2020-01-04:
For Linux, you need to have udev rules to gain access to the device stream. See the [FAQs](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception)

InfinityGhost, 2020-01-14:
General agreement that as of right now, the consistent fixes are located in the wiki. Closing but keeping this in mind for future improvements.

closed 2020-01-14 by InfinityGhost