OpenTabletDriver/OpenTabletDriver #61: Abnormal CPU usage when tablet is (dis)connected after startup

MoeLava, 2019-12-17:
## Description

After opening OpenTabletDriver without a tablet connected, manually connecting it afterwards will cause the CPU usage to jump to 100% and persist until closed.
Opening OpenTabletDriver with a tablet connected and disconnecting the tablet afterwards will also cause the same issue.
[image]

## System Information:

Option            | Value
------------------|------------------
Operating System  | Arch Linux
Software Version  | 0.1.0-4eff49e4
Tablet            | Gaomon S620

InfinityGhost, 2019-12-18:
This is likely an issue embedded in the HidSharp library, and I'm going to look for a proper fix if I can.

maxerium, 2020-01-24:
Did some pinpointing on this issue, and found out that it also happens on other HID devices (keyboards, mice, etc.) - CPU usage still ramps up. On other USB devices nothing happens, usage stays normal.

## System Info:
Setting | Value
----- | -----
Distro | Manjaro (5.4.12-xanmod7 kernel)
Driver Version | 0.1.0-8f72f61
Tablet | Wacom CTL-470

InfinityGhost, 2020-01-24:
Reproduced it and followed the trace here, it seems to end up on an endless retry loop when a device disconnects.
https://streamable.com/ftgji

maxerium, 2020-01-24:
Commit a8b85decb62759dc4c840d21107a324b4fc12d95 fixes this - did some testing. Excited for 0.1.1.

PR #87 (2020-01-25): Use HidSharpCore library
# Changes
- Use custom HIDSharp library with patches (https://github.com/InfinityGhost/HIDSharpCore)

## Issues
- Resolves #61
closed 2020-01-25 by InfinityGhost