OpenTabletDriver/OpenTabletDriver #68: Add the need to unload the "wacom" driver to the Linux wiki

Kodehawa, 2019-12-31:
## Description

For the best experience, specially on dual screen systems, I noticed unloading the "wacom" driver gives a way, way smoother experience and avoids the "ghost cursor" that appears on the other screen. 
Basically `sudo rmmod wacom`.
I just made a small bash script to run before OpenTabletDriver but I guess it could be added to the wiki.

## System Information:

Option            | Value
------------------|------------------
Operating System  |  Linux
Software Version  |  latest
Tablet            |  CTL-480

InfinityGhost, 2020-01-01:
Noted in the [wiki](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#i-have-a-ghost-cursor-constantly-alternating-cursor-position)

InfinityGhost, 2020-01-01:
Closing until further notice

closed 2020-01-01 by InfinityGhost