OpenTabletDriver/OpenTabletDriver #82: Deco 01 v2 do not respond

pomidorfarm, 2020-01-16:
## Description

I read Wiki, so I installed xp-pen driver but Cursor do not move.
would you help me ?
(already changed ProductID and ReportLength)
[image]
[image]
[image]
## System Information:

Option            | Value
------------------|------------------
Operating System  | Windows 10
Software Version  | v0.1.0-rc1
Tablet            | XP-Pen Deco 01 v2

InfinityGhost, 2020-01-16:
Just to make sure, the xp-pen driver is closed, correct?

pomidorfarm, 2020-01-16:
I closed driver at task manager and disable startup options

InfinityGhost, 2020-01-18:
I don't see any reason why it shouldn't be positioning your cursor. Try [this build](https://github.com/InfinityGhost/OpenTabletDriver/suites/399041841/artifacts/1105666) and you might have better luck

pomidorfarm, 2020-01-18:
Not moving.
I tried ↓
deleted Vmulti driver.
installed xp-pen driver,.NET Core 3.0 SDK.
closed xp-pen driver at task manager .
run opentabletdriver ,enable driver.
The tablet is sensing input, but the cursor does not move.
[image]

InfinityGhost, 2020-01-22:
Once plugin support is added I can better help treat this issue easier without direct contact. For now, the best option to help me mitigate this is to join my [discord](https://discord.gg/P3GYdUS)

InfinityGhost, 2020-01-22:
Actually scratch that, go into the configuration manager and modify your Deco 01 v2 config to match this, then close the window. It should begin to function after that.
 
[image]

pomidorfarm, 2020-01-22:
It worked , Thank you very much!!!
I joined discord.
Thank you!!!

InfinityGhost, 2020-01-22:
Reopening this as a reminder to fix the configuration in a future update.

InfinityGhost, 2020-02-03:
I'm pretty sure I fixed this tablet configuration, so closing this issue.

closed 2020-02-03 by InfinityGhost