OpenTabletDriver/OpenTabletDriver #109: [Detect:Error] Failed to get device input report length on XP-PEN G640

CaptainBasculin, 2020-03-25:
## Description

Tablet worked pretty much plug and play, but when I run the tablet driver, I get [Detect:Error] Failed to get device input report length. Visit the wiki (https://github.com/InfinityGhost/OpenTabletDriver/wiki) for more information. The tablet still moves, but absolutely unrelated to the driver. 

I've waited some time, and recorded everything on /var/log/Xorg.0.log. These are the tablet related outputs it gave me, in case that it helps I'm pasting it below. 

[    53.567] (II) config/udev: Adding input device XP-PEN STAR G640 Mouse (/dev/input/event8)
[    53.568] (**) XP-PEN STAR G640 Mouse: Applying InputClass "libinput pointer catchall"
[    53.568] (II) Using input driver 'libinput' for 'XP-PEN STAR G640 Mouse'
[    53.568] (**) XP-PEN STAR G640 Mouse: always reports core events
[    53.568] (**) Option "Device" "/dev/input/event8"
[    53.568] (**) Option "_source" "server/udev"
[    53.628] (II) event8  - XP-PEN STAR G640 Mouse: is tagged by udev as: Mouse
[    53.628] (II) event8  - XP-PEN STAR G640 Mouse: device is a pointer
[    53.628] (II) event8  - XP-PEN STAR G640 Mouse: device removed
[    53.657] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-4/1-4:1.0/0003:28BD:0094.0003/input/input14/event8"
[    53.657] (II) XINPUT: Adding extended input device "XP-PEN STAR G640 Mouse" (type: MOUSE, id 12)
[    53.657] (**) Option "AccelerationScheme" "none"
[    53.657] (**) XP-PEN STAR G640 Mouse: (accel) selected scheme none/0
[    53.657] (**) XP-PEN STAR G640 Mouse: (accel) acceleration factor: 2.000
[    53.657] (**) XP-PEN STAR G640 Mouse: (accel) acceleration threshold: 4
[    53.720] (II) event8  - XP-PEN STAR G640 Mouse: is tagged by udev as: Mouse
[    53.720] (II) event8  - XP-PEN STAR G640 Mouse: device is a pointer
[    53.722] (II) config/udev: Adding input device XP-PEN STAR G640 Mouse (/dev/input/mouse1)
[    53.722] (II) No input driver specified, ignoring this device.
[    53.722] (II) This device may have been added with another device file.
[    53.723] (II) config/udev: Adding input device XP-PEN STAR G640 Mouse (/dev/input/event14)
[    53.723] (**) XP-PEN STAR G640 Mouse: Applying InputClass "libinput pointer catchall"
[    53.723] (II) Using input driver 'libinput' for 'XP-PEN STAR G640 Mouse'
[    53.723] (**) XP-PEN STAR G640 Mouse: always reports core events
[    53.723] (**) Option "Device" "/dev/input/event14"
[    53.723] (**) Option "_source" "server/udev"
[    53.727] (II) event14 - XP-PEN STAR G640 Mouse: is tagged by udev as: Mouse
[    53.727] (II) event14 - XP-PEN STAR G640 Mouse: device is a pointer
[    53.727] (II) event14 - XP-PEN STAR G640 Mouse: device removed
[    53.765] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-4/1-4:1.0/0003:28BD:0094.0003/input/input15/event14"
[    53.765] (II) XINPUT: Adding extended input device "XP-PEN STAR G640 Mouse" (type: MOUSE, id 13)
[    53.765] (**) Option "AccelerationScheme" "none"
[    53.765] (**) XP-PEN STAR G640 Mouse: (accel) selected scheme none/0
[    53.765] (**) XP-PEN STAR G640 Mouse: (accel) acceleration factor: 2.000
[    53.765] (**) XP-PEN STAR G640 Mouse: (accel) acceleration threshold: 4
[    53.769] (II) event14 - XP-PEN STAR G640 Mouse: is tagged by udev as: Mouse
[    53.769] (II) event14 - XP-PEN STAR G640 Mouse: device is a pointer
[    53.771] (II) config/udev: Adding input device XP-PEN STAR G640 Mouse (/dev/input/mouse4)
[    53.771] (II) No input driver specified, ignoring this device.
[    53.771] (II) This device may have been added with another device file.
[    53.772] (II) config/udev: Adding input device XP-PEN STAR G640 Pen (/dev/input/event9)
[    53.772] (**) XP-PEN STAR G640 Pen: Applying InputClass "libinput tablet catchall"
[    53.772] (II) Using input driver 'libinput' for 'XP-PEN STAR G640 Pen'
[    53.772] (**) XP-PEN STAR G640 Pen: always reports core events
[    53.772] (**) Option "Device" "/dev/input/event9"
[    53.772] (**) Option "_source" "server/udev"
[    53.836] (II) event9  - XP-PEN STAR G640 Pen: is tagged by udev as: Tablet
[    53.976] (II) event9  - XP-PEN STAR G640 Pen: tablet 'XP-PEN STAR G640 Pen' unknown to libwacom
[    53.977] (II) event9  - XP-PEN STAR G640 Pen: device is a tablet
[    53.977] (II) event9  - XP-PEN STAR G640 Pen: device removed
[    54.005] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-4/1-4:1.1/0003:28BD:0094.0004/input/input16/event9"
[    54.005] (II) XINPUT: Adding extended input device "XP-PEN STAR G640 Pen" (type: TABLET, id 14)
[    54.072] (II) event9  - XP-PEN STAR G640 Pen: is tagged by udev as: Tablet
[    54.074] (II) event9  - XP-PEN STAR G640 Pen: tablet 'XP-PEN STAR G640 Pen' unknown to libwacom
[    54.074] (II) event9  - XP-PEN STAR G640 Pen: device is a tablet
[    54.076] (II) config/udev: Adding input device XP-PEN STAR G640 Pen (/dev/input/mouse2)
[    54.076] (II) No input driver specified, ignoring this device.
[    54.076] (II) This device may have been added with another device file.

[  1260.014] (II) libinput: XP-PEN STAR G640 Pen: needs a virtual subdevice
[  1260.014] (**) XP-PEN STAR G640 Pen Pen (0): Applying InputClass "libinput tablet catchall"
[  1260.014] (II) Using input driver 'libinput' for 'XP-PEN STAR G640 Pen Pen (0)'
[  1260.014] (**) XP-PEN STAR G640 Pen Pen (0): always reports core events
[  1260.014] (**) Option "Device" "/dev/input/event9"
[... 50 lines cut ...]
[ 11255.745] (**) XP-PEN STAR G640 Pen: Applying InputClass "libinput tablet catchall"
[ 11255.745] (II) Using input driver 'libinput' for 'XP-PEN STAR G640 Pen'
[ 11255.745] (**) XP-PEN STAR G640 Pen: always reports core events
[ 11255.745] (**) Option "Device" "/dev/input/event8"
[ 11255.745] (**) Option "_source" "server/udev"
[ 11255.812] (II) event8  - XP-PEN STAR G640 Pen: is tagged by udev as: Tablet
[ 11255.826] (II) event8  - XP-PEN STAR G640 Pen: tablet 'XP-PEN STAR G640 Pen' unknown to libwacom
[ 11255.826] (II) event8  - XP-PEN STAR G640 Pen: device is a tablet
[ 11255.826] (II) event8  - XP-PEN STAR G640 Pen: device removed
[ 11255.877] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-3/1-3:1.1/0003:28BD:0094.0002/input/input19/event8"
[ 11255.877] (II) XINPUT: Adding extended input device "XP-PEN STAR G640 Pen" (type: TABLET, id 13)
[ 11255.944] (II) event8  - XP-PEN STAR G640 Pen: is tagged by udev as: Tablet
[ 11255.946] (II) event8  - XP-PEN STAR G640 Pen: tablet 'XP-PEN STAR G640 Pen' unknown to libwacom
[ 11255.946] (II) event8  - XP-PEN STAR G640 Pen: device is a tablet
[ 11255.947] (II) config/udev: Adding input device XP-PEN STAR G640 Pen (/dev/input/mouse1)
[ 11255.947] (II) No input driver specified, ignoring this device.
[ 11255.947] (II) This device may have been added with another device file.

## System Information:

Option            | Value
------------------|------------------
Operating System  | Pop OS 19.10 (DE: KDE Plasma 5.16.5)
Software Version  | 0.1.1-rc1
Tablet            | XPPEN G640

MoeLava, 2020-03-25:
Can you try `sudo rmmod hid-uclogic`?

CaptainBasculin, 2020-03-25:
Done, after restarting the driver it gives
[Normal:Detect] Using report parser type 'TabletDriverLib.Tablet.TabletReportParser'.
[Normal:Settings]       Using output mode 'Absolute'
[Normal:Settings]       Set display area: [1366x768@683,384:0°],
[Normal:Settings]       Set tablet area:  [80x44.978035@60.639336,41.355103:0°],
[Normal:Settings]       Clipping is disabled
[Normal:Settings]       Tip Binding: 'None'@0%
[Normal:Settings]       Pen Bindings: [0, None], [1, None]
[Normal:Settings]       Express Key Bindings: [0, None], [1, None], [2, None], [3, None]
[Normal:Settings]       Applied all settings.
[Normal:Settings]       Using output mode 'Absolute'
[Normal:Settings]       Set display area: [1366x768@683,384:0°],
[Normal:Settings]       Set tablet area:  [80x44.978035@60.639336,41.355103:0°],
[Normal:Settings]       Clipping is disabled
[Normal:Settings]       Tip Binding: 'None'@0%
[Normal:Settings]       Pen Bindings: [0, None], [1, None]
[Normal:Settings]       Express Key Bindings: [0, None], [1, None], [2, None], [3, None]
[Normal:Settings]       Applied all settings.

but still doesn't affect the way tablet works. It's still on full area, which it always has been

InfinityGhost, 2020-03-25:
Run `lsusb -v` and return output here

InfinityGhost, 2020-03-25:
After that, do `sudo cat /sys/kernel/debug/usb/devices` and return output here

CaptainBasculin, 2020-03-25:
[Output of lsusb -v](https://github.com/InfinityGhost/OpenTabletDriver/files/4380911/lsusb.-v.txt)

[Output of sudo cat ⁄sys⁄kernel⁄debug⁄usb⁄devices](https://github.com/InfinityGhost/OpenTabletDriver/files/4380913/sudo.cat.sys.kernel.debug.usb.devices.txt)

InfinityGhost, 2020-03-25:
The `usbhid` module took over after you removed the `hid-uclogic` module, so now we need to remove it from that device only. If we were to rmmod the `usbhid` module you wouldn't have any input.

We need to work out something else instead.

InfinityGhost, 2020-03-31:
A similar case I have worked with by forcing modules was with my GPU VFIO setup, and this is a snippet of the script I have to use to bind the `nvidia` module after the virtual machine stops.
```bash
# PCI BUS ID
IOMMU_GPU="01:00.0"
videoid="10de 1b81"
videobusid="0000:$IOMMU_GPU"

echo $videoid > /sys/bus/pci/drivers/nvidia/new_id
echo $videobusid > /sys/bus/pci/devices/$videobusid/driver/unbind
echo $videobusid > /sys/bus/pci/drivers/nvidia/bind
echo $videoid > /sys/bus/pci/drivers/nvidia/remove_id
```
Perhaps we can do something similar on the USB bus? We already know the `idVendor` and `idProduct` of the device, we just need the bus ID. On my system, for example, the Wacom CTL-480 can be found at `/sys/bus/usb/devices/2-1.4.1` by searching through all of the devices for the matching vendor and product id in hexadecimal format. We might be able to find a solution for your case specifically if you can find a device under `/sys/bus/usb/devices/` that has `idVendor` equal to `28bd` and `idProduct` equal to `0094`.

CaptainBasculin, 2020-04-03:
such a device exists under /sys/bus/usb/devices/1-4/.

InfinityGhost, 2020-04-03:
```bash
#!/bin/bash
deviceid="28bd 0094"
devicebusid="1-4"

echo $deviceid > /sys/bus/usb/devices/$devicebusid/driver/unbind
```
This script might work, requires running as root

GabD13, 2020-04-24:
I'm having the same problems, but with my VEIKK S640, what could be the progress to fixing the issue?

InfinityGhost, 2020-05-05:
Still haven't had anyone try the fix I mentioned so I'm closing this until someone else brings this to light

closed 2020-05-05 by InfinityGhost

nobbele, 2020-05-23:
Not sure what it's supposed to do exactly but I tried running it and it changed nothing (g430s)

cartals, 2020-06-16:
@nobbele I have g430s too, I have blacklisted hid-uclogic. My tablet's id is 28bd 0075 and if I don't remeber wrong I have found unbind's location at ```/sys/bus/usb/devices/3-4/driver``` but sadly I couldn't change unbind and could not use echo command despite I did it on root user.

InfinityGhost, 2020-06-18:
> @nobbele I have g430s too, I have blacklisted hid-uclogic. My tablet's id is 28bd 0075 and if I don't remeber wrong I have found unbind's location at `/sys/bus/usb/devices/3-4/driver` but sadly I couldn't change unbind and could not use echo command despite I did it on root user.

We still haven't found a proper workaround for the G640, its fairly broken as of right now. The only thing I can suggest is applying the libinput override to your udev rules
https://github.com/InfinityGhost/OpenTabletDriver-udev/wiki