OpenTabletDriver/OpenTabletDriver #115: Gaomon s620 not found

Zothin, 2020-04-03:
the driver searched for tablets and didn't find my s620 I tried researching and it failed again.

InfinityGhost, 2020-04-03:
Logs? Platform? Version?

Zothin, 2020-04-03:
Latest Version 0.3.1 on Windows 10 
[4-4-2020.log](https://github.com/InfinityGhost/OpenTabletDriver/files/4430250/4-4-2020.log)

InfinityGhost, 2020-04-03:
Open OpenTabletDriver, go to the configuration manager under the Tablets menu. In the configuration manager there is a devices tab. Send a screenshot of the contents of the tab.

Zothin, 2020-04-03:
http://prntscr.com/rsm98d

Zothin, 2020-04-03:
as far as i can see in my device registers my s620 as a Gaomon Tablet with a maximum Input report length of 10. In the devices the s620 has a 12 IRL and the vendor id and product id are identical if this is of any help

InfinityGhost, 2020-04-03:
Lets try a newer build to see if this is fixed already. https://github.com/InfinityGhost/OpenTabletDriver/suites/560137891/artifacts/3635112

Zothin, 2020-04-03:
kk

Zothin, 2020-04-03:
It's 2 am here and the fact that you replied and took the time to help me instantly is commendable. Thanks.

Zothin, 2020-04-03:
Still no tablet found.

Zothin, 2020-04-03:
Wait i changed the IRL in the configurator and this happened
[2.15 4-4-2020.log](https://github.com/InfinityGhost/OpenTabletDriver/files/4430318/2.15.4-4-2020.log)

InfinityGhost, 2020-04-03:
Do you happen to have the Gaomon driver installed?

Zothin, 2020-04-03:
No I had your hawku version before but windows kept detecting a virus so i just got otd

InfinityGhost, 2020-04-03:
Uninstall the vmulti drivers and reboot

Zothin, 2020-04-03:
On it

Zothin, 2020-04-03:
Where does one find the vmulti drivers?

InfinityGhost, 2020-04-03:
It's in the releases of hawku's tabletdriver, https://github.com/hawku/TabletDriver/releases/download/v0.2.3/TabletDriverV0.2.3.zip

Zothin, 2020-04-03:
Found it and im rebooting

Zothin, 2020-04-03:
I have succesfully rebooted and launched otd with no success

InfinityGhost, 2020-04-06:
Send a screenshot of the devices tab and any of the Gaomon Tablet devices' pages

Zothin, 2020-04-06:
http://prntscr.com/rtxa1i
http://prntscr.com/rtxaam
Here they are

InfinityGhost, 2020-04-06:
Modify the `Configurations/Gaomon/S620.json` file to contain this: 
```json
{
  "Name": "Gaomon S620",
  "VendorID": 9580,
  "ProductID": 109,
  "InputReportLength": 12,
  "ReportParser": "TabletDriverLib.Vendors.Gaomon.GaomonReportParser",
  "CustomInputReportLength": 10,
  "CustomReportParser": "TabletDriverLib.Vendors.Gaomon.GaomonReportParser",
  "Width": 165.1,
  "Height": 101.6,
  "MaxX": 33020.0,
  "MaxY": 20320.0,
  "MaxPressure": 8191,
  "ActiveReportID": 10,
  "AuxReportLength": 0,
  "AuxReportParser": null,
  "FeatureInitReport": null
}
```

InfinityGhost, 2020-04-10:
This issue may be the result of a dirty installation it turns out, most tablet driver uninstallers don't clean up like they should and lead to improper initialization of the device.

doubledge-spec, 2020-04-18:
I tried all of the things mentioned in this thread, still doesn't detect s620

doubledge-spec, 2020-04-18:
> I tried all of the things mentioned in this thread, still doesn't detect s620

I had to uninstall the Gaomon drivers for it to detect, seems kinda backwards considering that Your Tablet Driver fork from hawku needs it.

InfinityGhost, 2020-04-30:
> > I tried all of the things mentioned in this thread, still doesn't detect s620
> 
> I had to uninstall the Gaomon drivers for it to detect, seems kinda backwards considering that the Tablet Driver fork from hawku needs it.

I'll consider this the fix then.

closed 2020-04-30 by InfinityGhost