OpenTabletDriver/OpenTabletDriver #122: Pressing the pen onto the tablet stops mouse movement until the pen is lifted again

kangalioo, 2020-04-28:
## Description
Pressing the pen onto the tablet stops mouse movement until the pen is lifted again. I don't even know if it's a feature or a bug, because I suppose this is useful for people who use their pen like a computer mouse. It's unwanted behavior for many other use cases though.

It would be good to be able to toggle this behavior.

## System Information:

Option            | Value
------------------|------------------
Operating System  | Arch (Antergos) Linux x64, KDE
Software Version  | v0.2.0-rc1
Tablet            | Wacom PTH-451

InfinityGhost, 2020-04-29:
We're gonna need some capture of the tablet debugger between when the pen is pressing and when its not.

kangalioo, 2020-04-29:
I recorded a video of the Tablet Debugger here: https://streamable.com/1h7zez

You can see whether I'm currently pressing or hovering from the `Pressure` property.

InfinityGhost, 2020-04-30:
Try this build
https://github.com/InfinityGhost/OpenTabletDriver/suites/641017376/artifacts/5359286

kangalioo, 2020-04-30:
It works perfectly as far as I can tell. Not sure if I should close this issue now

InfinityGhost, 2020-04-30:
Don't close the issue manually, the pull request merge will close it.

PR #125 (2020-04-30): Fix minimum report ID test
# Changes

- [x] Fix report ID test in all output modes

# Issues
- Resolves #122
closed 2020-04-30 by InfinityGhost