OpenTabletDriver/OpenTabletDriver #128: It can't detect XP-PEN G430S

cartals, 2020-05-04:
## Description
I started driver and clicked "detect" but it doesn't work, terminal says "Can't detect"

## System Information:

Option            | Value
------------------|------------------
Operating System  | Manjaro XFCE 20
Software Version  | v0.2.0 (aur version)
Tablet            | XP-PEN G430S

InfinityGhost, 2020-05-04:
Copy the diagnostic dump and send it here, preferably through pastebin

cartals, 2020-05-04:
Actual Pastebin banned on my country so I posted on ubuntu pastebin.
https://paste.ubuntu.com/p/Ms3kcPBdfB/

InfinityGhost, 2020-05-04:
https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#notice-for-uclogic-tablet-owners
`sudo rmmod hid-uclogic`

cartals, 2020-05-04:
sudo rmmod hid-uclogic I did it and retried but it didn't work

InfinityGhost, 2020-05-04:
Resend dump after the detection fix

cartals, 2020-05-04:
It doesn't give dumps with sudo, but If I would use it with sudo then it wouldn't detect tablet, Whar can I do?

InfinityGhost, 2020-05-04:
```bash
sudo groupadd users
sudo usermod -a -G users $USER
```
Re-login, this aught to fix the permission problem so you can run as user and provide at least console output

cartals, 2020-05-04:
It can detect right now but it is still doesn't give dumps. 
`/usr/bin/opentabletdriver: line 3: 34471 Segmentation fault      (core dumped) dotnet OpenTabletDriver.dll "$@"`

InfinityGhost, 2020-05-04:
Exporting the dump for whatever reason fails because it can't open the file dialog, known issue. It shouldn't dump core if you copy the diagnostic dump

cartals, 2020-05-04:
It does not work too, also should I do this sudo usermod -a -G users $USER everytime I log out it was working but second time it wasn't detecting again.

cartals, 2020-05-05:
It works now but I mean it doesn't feel smooth?

InfinityGhost, 2020-05-05:
It won't be smooth without a filter applied. OpenTabletDriver takes the raw data output from your tablet otherwise.

cartals, 2020-05-05:
Okay got it but when I click filters section it becomes blank.

InfinityGhost, 2020-05-05:
https://github.com/InfinityGhost/OpenTabletDriver/releases/download/v0.2.0-rc1/OpenTabletDriverPlugins.dll
https://github.com/InfinityGhost/OpenTabletDriver/wiki/Plugins#plugin-installation

closed 2020-05-05 by cartals