OpenTabletDriver/OpenTabletDriver #131: Can't use my tablet in OneNote + no pen pressure in Photoshop

shader301202, 2020-05-13:
## Description
Tablet works fine in Osu. I cannot use it in OneNote or Windows Ink Sketchpad as it detects it as mouse? and I cannot draw. It works in Photoshop except there is no pen pressure. How can I fix this? Or is it by design (that the drivers only work in osu)?

## System Information:
Option            | Value
------------------|------------------
Operating System  | Windows 10 x64
Software Version  | 0.2.0-rc1
Tablet            | Gaomon S620

InfinityGhost, 2020-05-13:
Windows Ink has to be added manually in order to support pen pressure. It's in the plans to add it but as of right now its not supported.

closed 2020-05-13 by InfinityGhost

shader301202, 2020-05-14:
Alright, thanks for the answer! :D  

I will follow the releases while waiting for the feature ;p

InfinityGhost, 2020-05-15:
I'd suggest following [this project](https://github.com/InfinityGhost/OpenTabletDriverPlugins) in that case, however it will only be published as binaries with future releases.