OpenTabletDriver/OpenTabletDriver #132: Star G640 Not Detected, Possibly new revision

skyborgff, 2020-05-13:
## Description
Star G640 Isn't detected, aparently its a new revisio
[image]
[image]

## System Information:
Operating System  Windows 10
Software Version  v0.2.0-rc1
Tablet            G650

InfinityGhost, 2020-05-13:
Export a diagnostic report and send it here. Use [this](https://github.com/InfinityGhost/OpenTabletDriver/suites/660195867/artifacts/5800779) version which has diagnostic report support.

InfinityGhost, 2020-05-13:
Input Report Length: 12
Still unsure if this is a different revision or something though

closed 2020-05-15 by InfinityGhost