OpenTabletDriver/OpenTabletDriver #136: Configuration overall does nothing (gaomon s620)

Mrcubix, 2020-05-15:
## Description
Configuration overall does nothing.
uninstalled the OEM drivers, rebooted, the tablet work, this tool detect my tablet, but when i edit anything, it just doesn't work after applying them.

## System Information:

Option            | Value
------------------|------------------
Operating System  | Windows 8.1 x64
Software Version  | Latest as of this (0.20--rc1)
Tablet            |  Gaomon s620

ssccoonniicc, 2020-05-15:
Gaomon s620 support on windows is pretty hit **or** miss. OTD _in most cases_ does not need a driver to be installed for it to work. If the vmulti driver is still installed then try to uninstall it. Also in OTD click on the tablets button on the top and then open tablet debugger. Then move your pen around on your tablet and tell me if and thing changes in the debugger.

Mrcubix, 2020-05-15:
nothing happen in the debugger, no vmulti driver installed

skyborgff, 2020-05-15:
Just to clarify, it detects, and moves, but changing the areas doesnt "apply"?

Mrcubix, 2020-05-15:
There is the msg about the setting getting applied, but nothing change on the tablet

Mrcubix, 2020-05-15:
ok, now it work for some reason on desktop, but it doesn't work ingame in osu

Vlothx, 2020-05-15:
Please disable raw input.

Mrcubix, 2020-05-15:
already disabled

InfinityGhost, 2020-05-15:
> EDIT: worked after restart, thx, still it's strange that OTD started working randomly
> 
> other than this, is there any way to make OTD work in background?

Not right now, but after #130 is merged it'll be the default.

closed 2020-05-15 by InfinityGhost