OpenTabletDriver/OpenTabletDriver #147: cursor glitched

a0g-402, 2020-06-03:
## Description

the cursor sometime glitch
i cant fc any song ;(
[https://youtu.be/fHUvO4to5hk](url)
## System Information:

Option            | Value
------------------|------------------
Operating System  | Windows 10 Home
Software Version  | 0.2.0--rc1
Tablet            | Wacom CTL-471

Vlothx, 2020-06-03:
At least on the video you sent, looks like there's another driver running in the background.

InfinityGhost, 2020-06-03:
Try [this build](https://github.com/InfinityGhost/OpenTabletDriver/suites/737612381/artifacts/7454795) to see if the problem still occurs.

a0g-402, 2020-06-04:
> Try [this build](https://github.com/InfinityGhost/OpenTabletDriver/suites/737612381/artifacts/7454795) to see if the problem still occurs.

I tried, problem still occurs but more rarely
I will try other drivers

a0g-402, 2020-06-04:
ok i know what caused it sorry for wasting your time

closed 2020-06-04 by a0g-402