OpenTabletDriver/OpenTabletDriver #148: Gaomon S620 Report Rate

r3x6k, 2020-06-04:
## Description

Using OpenTabletDriver I get constant 144hz on zowie's mouse-rate-checker website (https://zowie.benq.com/ja/support/mouse-rate-checker.html). When using either the default Gaomon drivers or the fork of hawku's TabletDriver, I get around 266hz. Unsure if this is intentional for OTD or bugged on my end. This hasn't really affected my gameplay (maybe better/faster osu! players might notice it), but still curious on why this happens.

## System Information:

Option            | Value
------------------|------------------
Operating System  | Windows 10 
Software Version  | OTD v0.2.0-rc1
Tablet            | Gaomon S620

InfinityGhost, 2020-06-04:
There's nothing directly programmed into OTD that would result in a lower report rate, unless the HidSharpCore library we use somehow rate limits reports. As far as I'm aware though it doesn't. Most users get the normal 266hz.

r3x6k, 2020-06-05:
Alright then, probably just some issue with my computer :(. OTD still works fine though, so I'll just stick with it and if any troubles come up I'll switch back to your fork of hawku's TabletDriver.

closed 2020-06-05 by r3x6k