OpenTabletDriver/OpenTabletDriver #159: Hey there I'm using an XP PEN G640 tablet, doesn't seem to work for me :/

17steen, 2020-06-25:
I don't know if the fact that I have this tablet can be any helpful to you, I appreciate that you are working on this driver and wish you the best through this process.

I have installed the digimend drivers before trying this one out, maybe they are mutually incompatible?

(Kind of new to this git hub thing, not sure if this is the right place)

TOJ0, 2020-06-25:
My guess is that you have the revision b tablet (second generation) which means you have to change in your config file to make it detect the tablet. I recommend you join the OTD discord and ask there for more in-depth help.

InfinityGhost, 2020-06-25:
1. What platform are you running (including distribution if on linux)
2. What version of OTD are you running

closed 2020-06-28 by 17steen