OpenTabletDriver/OpenTabletDriver #167: OTD does not detect Gaomon S620 under Pop!OS 20.04 LTS

PowerPixel, 2020-07-07:
## Description
Hello,
i tried to run OTD under my installation of PopOS two ways : 
- The first one, through the official release, 2.0-RC1, with the .deb package. However, i wasn't able to get my tablet to be detected, so i headed to the discord for some help with troubleshooting. I tried disabling hid-uclogic module, but unfortunately, it just stops every USB device plugged into my computer, so i can't really do much.

- The second one, through the artifact linked below, which didn't worked either. First, i had to run the daemon with sudo, but it always spat out this error : 
```
[Normal:Detect]	Searching for tablet 'Gaomon S620'
[Error:System.ArgumentOutOfRangeException]	Value range is [0, 15]. (Parameter 'value')
   at HidSharp.Reports.Units.Unit.DecodeExponent(UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseMainData(MainItemTag tag, UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseMain(MainItemTag tag, UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItem(EncodedItem item)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItems(IEnumerable`1 items)
   at HidSharp.Reports.ReportDescriptor.ParseRawReportDescriptor(Byte[] buffer)
   at HidSharp.Reports.ReportDescriptor..ctor(Byte[] buffer)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at TabletDriverLib.Driver.<>c__DisplayClass25_0.<TryMatchTablet>b__2(HidDevice device) in /home/runner/work/OpenTabletDriver/OpenTabletDriver/TabletDriverLib/Driver.cs:line 90
   at System.Linq.Utilities.<>c__DisplayClass1_0`1.<CombinePredicates>b__0(TSource x)
   at System.Linq.Utilities.<>c__DisplayClass1_0`1.<CombinePredicates>b__0(TSource x)
   at System.Linq.Enumerable.WhereEnumerableIterator`1.GetCount(Boolean onlyIfCheap)
   at System.Linq.Enumerable.Count[TSource](IEnumerable`1 source)
   at TabletDriverLib.Driver.TryMatchTablet(TabletProperties tablet, HidDevice& tabletDevice, IReportParser`1& parser) in /home/runner/work/OpenTabletDriver/OpenTabletDriver/TabletDriverLib/Driver.cs:line 94
   at TabletDriverLib.Driver.TryMatch(TabletProperties tablet) in /home/runner/work/OpenTabletDriver/OpenTabletDriver/TabletDriverLib/Driver.cs:line 41
[Error:Detect]	Failed to get device.https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#notice-for-uclogic-tablet-owners
```

Launching the GTK, i tried to detect my tablet, but it spat out this error : 
`[Normal:Detect]	Configuration directory '/home/myuname/Téléchargements/OTD/OpenTabletDriver.Daemon/bin/Release/netcoreapp3.1/linux-x64/publish/Configurations' does not exist.
`
even though my Configurations folder exists at this path.

## System Information:
Operating System  | Pop!OS 20.04 LTS
Software Version  | https://github.com/InfinityGhost/OpenTabletDriver/suites/874418404/artifacts/10205061
Tablet            | Gaomon S620

InfinityGhost, 2020-07-07:
> Launching the GTK, i tried to detect my tablet, but it spat out this error :
> `[Normal:Detect] Configuration directory '/home/myuname/Téléchargements/OTD/OpenTabletDriver.Daemon/bin/Release/netcoreapp3.1/linux-x64/publish/Configurations' does not exist.`

Put this in a separate issue.

InfinityGhost, 2020-07-07:
This issue matches #160, however I won't close as duplicate because it covers it for another tablet that should be unaffected based on that issue.

InfinityGhost, 2020-07-11:
Resolved, turned out to be the DE. It's still odd that the libinput override is resulting in these exceptions.

closed 2020-07-11 by InfinityGhost