OpenTabletDriver/OpenTabletDriver #169: GTK can't find Configurations folder at path, even if it exists.

PowerPixel, 2020-07-08:
## Description

Launching the GTK, i tried to detect my tablet, but it spat out this error :

`[Normal:Detect] Configuration directory '/home/myuname/Téléchargements/OTD/OpenTabletDriver.Daemon/bin/Release/netcoreapp3.1/linux-x64/publish/Configurations' does not exist.`

even though my Configurations folder exists at this path.

Please note, i won't be unable to reproduce this issue, since i recently switched to Manjaro linux.

## System Information:

Option            | Value
------------------|------------------
Operating System  |  Pop!OS 20.04 LTS
Software Version  |  https://github.com/InfinityGhost/OpenTabletDriver/suites/874418404/artifacts/10205061
Tablet            |  Gaomon S620

InfinityGhost, 2020-07-09:
Will try to reproduce this, but all I can figure is that this if block doesn't return a tablet
https://github.com/InfinityGhost/OpenTabletDriver/blob/master/OpenTabletDriver.UX/MainForm.cs#L410

InfinityGhost, 2020-07-09:
I've was unable to reproduce in a Debian 10 VM, so I really don't know if this is just a one off issue in that artifact.

closed 2020-07-10 by InfinityGhost