OpenTabletDriver/OpenTabletDriver #170: OTD does not detect Gaomon s620

doubledge-spec, 2020-07-08:
## Description

The driver does not detect the s620 but detects a device "Gaomon Tablet". both have identical vendor and product id's.
[image]
[image]
[image]

## System Information:

Option            | Value
------------------|------------------
Operating System  | Ubuntu
Software Version  | 20.04 LTS
Tablet            | Gaomon s620

doubledge-spec, 2020-07-08:
same issue I had with Windows 10 a couple of months ago. This is a fresh install of Ubuntu 20.04 LTS.

ssccoonniicc, 2020-07-08:
add `blacklist hid-uclogic` to /etc/modprobe.d/blacklist.conf, download [this file](https://cdn.discordapp.com/attachments/723399565780189234/729075215581184022/30-opentabletdriver.rules) and put it in /etc/udev/rules.d/ and then restart your computer.

doubledge-spec, 2020-07-08:
yeah I am new to linux, but I can figure this out thanks @ssccoonniicc

InfinityGhost, 2020-07-09:
Marking closed based on last reply.

closed 2020-07-09 by InfinityGhost