OpenTabletDriver/OpenTabletDriver #195: CTL-4100 is not actually supported

Remyria, 2020-07-17:
## Description

Installed, didn't work properly, cursor just did whatever.
Tried some things like restarting the driver, uninstalling and reinstalling vmulti from hawku in case it interfered, reinstalling OTD, changed the directory of bin, changed sensitivity, screen and tablet areas, NOTHING had the slightest impact on how well it worked.

## System Information:

Option            | Value
------------------|------------------
Operating System  |  W10 Family
Software Version  |  v0.3.0
Tablet            |  Wacom CTL-4100 (Intuos S Small)

InfinityGhost, 2020-07-17:
Moved the [card](https://github.com/InfinityGhost/OpenTabletDriver/projects/4#card-31065001) from Supported -> Nonfunctional

InfinityGhost, 2020-08-02:
Closing this issue because of #199, this issue moved the tablet from untested -> broken.

PR #237 (2020-08-03): Add IntuosV3 support
# Changes
- Add `IntuosV3Report`
  - Add `IntuosV3ReportParser`
  - Add `WacomDriverIntuosV3ReportParser`

## Tablets
Adds support for:
- Wacom CTL-4100
- Wacom CTL-6100
- Wacom PTH-660

# Issues
- Closes #195 
- Closes #199
closed 2020-08-02 by InfinityGhost