OpenTabletDriver/OpenTabletDriver #200: Add IGestureBinding

InfinityGhost, 2020-07-18:
## Description

This would provide a method to gesture using pen movement and a button, adding ability for things such as scrolling with the pen and so on.

InfinityGhost, 2021-04-20:
This is no longer needed as `IBinding` has been modified to pass the report itself. The report will contain the data needed in order to execute any sort of gesture.

closed 2021-04-20 by InfinityGhost

UlyssesZh, 2021-07-01:
So how to scroll with the pen now?

InfinityGhost, 2021-07-01:
A scrolling binding has yet to actually be implemented. This issue was tracking an feature which allowed for dynamic input to invoke a binding, not scrolling itself.

UlyssesZh, 2021-07-01:
> A scrolling binding has yet to actually be implemented. This issue was tracking an feature which allowed for dynamic input to invoke a binding, not scrolling itself.

Is there any plans that a scrolling binding will be added?

UlyssesZh, 2021-07-01:
> So how to scroll with the pen now?

I think binding scrolling is a frequently requested feature. It is also requested in #1158.

InfinityGhost, 2021-07-01:
It is still a work in progress, and I have a working technique on Linux. The issue is likely going to be the macOS implementation.

UlyssesZh, 2021-07-01:
> It is still a work in progress, and I have a working technique on Linux. The issue is likely going to be the macOS implementation.

What about Windows? I am a Windows user, and I want that feature of binding scrolling.

3ab3z2, 2025-03-26:
Is scrolling with the pen implemented on linux yet? (wayland)