OpenTabletDriver/OpenTabletDriver #211: Can't find OpenTabletDriver.exe in my folder

brakence, 2020-07-25:
I'm sorry in advance for being pretty clueless about all this stuff. I tried downloading the latest release of OpenTabletDriver so I could connect it to my Gaomon S620. I attempted to run OpenTabletDriver.Daemon.exe but only a text script appeared with nothing else happening.

X9VoiD, 2020-07-25:
OpenTabletDriver have two separate components to launch.

The first being the daemon, OpenTabletDriver.Daemon.exe

And then the UI,
`OpenTabletDriver.UX.Wpf` for Windows.
`OpenTabletDriver.UX.Gtk` for Linux.
`OpenTabletDriver.UX.MacOS` for MacOS.

InfinityGhost, 2020-07-25:
Assuming you're running Windows, I would suggest using the installer [here](https://github.com/InfinityGhost/OpenTabletDriver.Installer/releases/tag/v0.3.0.2). It handles updates and the two executables for you.

InfinityGhost, 2020-07-30:
No response in almost a week, assuming resolved.

closed 2020-07-30 by InfinityGhost

Airkek, 2020-08-24:
When I trying to start Daemon - it says 
`fail: JKang.IpcServiceFramework.Hosting.NamedPipe.NamedPipeIpcEndpoint[0]
      Error occurred
System.AggregateException: One or more errors occurred. (Some or all identity references could not be translated.)
 ---> System.Security.Principal.IdentityNotMappedException: Some or all identity references could not be translated.
   at System.Security.Principal.NTAccount.Translate(IdentityReferenceCollection sourceAccounts, Type targetType, Boolean forceSuccess)
   at System.Security.Principal.NTAccount.Translate(Type targetType)
   at System.Security.AccessControl.CommonObjectSecurity.ModifyAccess(AccessControlModification modification, AccessRule rule, Boolean& modified)
   at System.Security.AccessControl.CommonObjectSecurity.AddAccessRule(AccessRule rule)
   at JKang.IpcServiceFramework.Hosting.NamedPipe.NamedPipeIpcEndpoint`1.WaitAndProcessAsync(Func`3 process, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---`
[image]

Airkek, 2020-08-24:
[image]

InfinityGhost, 2020-08-24:
> When I trying to start Daemon - it says
> `fail: JKang.IpcServiceFramework.Hosting.NamedPipe.NamedPipeIpcEndpoint[0] Error occurred System.AggregateException: One or more errors occurred. (Some or all [...]

#243