OpenTabletDriver/OpenTabletDriver #243: System.AggregateException: One or more errors occurred. SpamTrain

Mrcubix, 2020-08-03:
## Description
This error get spammed in the console, causing high CPU usage, but also prevent the UI from connecting to the daemon
I've tried reinstalling the OTD driver from scratch, OEM drivers are killed, only 1 instance of OTD is started

[image]

https://pastebin.com/Y9UHKTCu

## System Information:

Option            | Value
------------------|------------------
Operating System  | Win 8.1 64x
Software Version  | ALL
Tablet            | Gaomon 620 (in my case)

InfinityGhost, 2020-08-12:
This may depend on https://github.com/jacqueskang/IpcServiceFramework/pull/156 being merged, along with a NuGet package update in order to fix.

InfinityGhost, 2020-08-24:
Now we're waiting on v3.0.5 of IpcServiceFramework to be released, which is currently in preview on NuGet.

Airkek, 2020-08-24:
[image]
Updating IpcServiceFramework to 3.0.5-preview helped

JaaceB, 2020-09-05:
Could someone pls tell me how to update to IpsFramework 3.0.5? I just cant get it working

Mrcubix, 2020-09-05:
> Could someone pls tell me how to update to IpsFramework 3.0.5? I just cant get it working

Replace the dll in OTD folder by this one : https://cdn.discordapp.com/attachments/615611007951306863/751201377249919036/JKang.IpcServiceFramework.Hosting.NamedPipe.dll

if the error still persist, then reboot your pc

JaaceB, 2020-09-05:
Now it says: Failed to open tablet. Make sure you have required permissions to open device streams.

Mrcubix, 2020-09-05:
> Now it says: Failed to open tablet. Make sure you have required permissions to open device streams.

i would suggest you come to the official discord server https://discord.gg/P3GYdUS we'll continue the discussion there

PR #283 (2020-09-11): Update IPC framework to v3.0.5 preview 1
# Issues
- Closes #243
closed 2020-09-11 by InfinityGhost