OpenTabletDriver/OpenTabletDriver #247: Add huion HS610 support

AbstractQbit, 2020-08-04:
## Description

VID is 0x256C, PID is 0x006D

Two endpoints: 0 (vendor-defined device) and 1 (pen)

Pen messages appear to be 14 bytes long (or 13? idk, zeros start at 13'th byte), first 12 of them are actual useful data.

Here is a little snippet of how I was reading data from it back when I played around with it in Feb: (can't believe it took me so long to stumble upon this project)
```
const char* DecodeState(uint16_t State)
{
    switch (State)
    {
    case 10: return "HoverStop";     // 0x 00 0A

    case 49162: return "Hover";      // 0x C0 0A
    case 49674: return "Hover + A";  // 0x C2 0A
    case 50186: return "Hover + B";  // 0x C4 0A

    case 49418: return "Touch";      // 0x C1 0A
    case 49930: return "Touch + A";  // 0x C3 0A
    case 50442: return "Touch + B";  // 0x C5 0A

    default: return "Unknown";
    }
}

struct PenInput
{
    PenInput(){}
    PenInput(UCHAR* buffer){ReadFromBuffer(buffer);}
    uint16_t State=0;
    uint16_t Xpos=0;
    uint16_t Ypos=0;
    uint16_t Pressure=0;
    int8_t Xtilt=0;
    int8_t Ytilt=0;

    void ReadFromBuffer(UCHAR* buffer)
    {
        State = *(uint16_t*)(buffer);
        Xpos = *(uint16_t*)(buffer + 2);
        Ypos = *(uint16_t*)(buffer + 4);
        Pressure = *(uint16_t*)(buffer + 6);
        Xtilt = *(int8_t*)(buffer + 8);
        Ytilt = *(int8_t*)(buffer + 9);
    }

    void Print()
    {
        printf("State=%05u (%-9s), Xpos=%05u, Ypos=%05u, Pressure=%05u, Xtilt=%03d, Ytilt=%03d\n", State, DecodeState(State), Xpos, Ypos, Pressure, Xtilt, Ytilt);
    }
};
```

Both X and Y pos axis go from 0 to 0x7FFF (7FFF), Pressure goes up to 8191, tilt axis reports are between -60 and 60 signed.

Report rate on spec is supposed to be 266 Hz, but with libusb I only could poll at 220 Hz, maybe I was doing something wrong.

InfinityGhost, 2020-08-04:
Biggest problem with adding support for Huion tablets in general is that they differentiate which tablet is which by a report descriptor string, which isn't supported *yet* in the HID library we use ([InfinityGhost/HIDSharpCore](https://github.com/InfinityGhost/HIDSharpCore)). We're currently working on this (https://github.com/InfinityGhost/HIDSharpCore/pull/1) and any additional information would be greatly appreciated.

AbstractQbit, 2020-08-04:
So huion uses the same pid for a bunch of different tablets, I didn't know that.

Here are string descriptors that I found:
```
String #0:
String #1:HUION
String #2:Huion Tablet
String #3:
String #4:
...
String #198:
String #199:
String #200:??|??????
String #201:HUION_T194_190307
String #202:HUION Animation Technology Co.,ltd
String #203:
String #204:
String #205:☺
String #206:
String #207:
...
String #254:
String #255:
```
I suppose the 201'st is the one needed to id the device

InfinityGhost, 2020-09-04:
Alright we can finally jump to working on this one now!

PR #277 (2020-09-07): Add Huion HS610 configuration
Not sure why I was getting 0-32767 ranges mentioned in #247 with libusb, in OTD tablet debugger x is 0-50800, y is 0-31750, so those were set.

Everything apart from ranges, dimensions, name and device string was copied from hs611.

Now OTD seems to work, so this closes #247.
closed 2020-09-07 by InfinityGhost