OpenTabletDriver/OpenTabletDriver #250: CTH-670 Bindings broken

acclaiming, 2020-08-08:
## Description

Tablet Tip binding doesn't work at all, binding the side buttons to backslash (Windows layout set to US-ANSI) outputs "["

Not sure if the "Filters" and "Tools" tabs are supposed to be empty

Operating System  |  Windows 10 Pro Version 2004
Software Version  | 0.3.2
Tablet            | Wacom CTH-670

acclaiming, 2020-08-08:
Just retested on Manjaro-XFCE: Pen tip works but I can't rebind any keys

ssccoonniicc, 2020-08-08:
> Not sure if the "Filters" and "Tools" tabs are supposed to be empty

Those tabs are empty because no plugins are installed.

> Raw Input in osu! doesn't work at all.

With the way OpenTabletDriver works right now raw input will not work

> Tablet Tip binding doesn't work at all, binding the side buttons to backslash (Windows layout set to US-ANSI) outputs "["

That might be a problem with the config

>  Edit2: it's not detecting the tablet even though I followed the guide in the linux faq, IDK what to do anymore, it worked for a 
second in the beginning but now I'm stumped

Make sure to blacklist the wacom module but if that doesn't work then you can join the discord server for some extra help: https://discord.gg/P3GYdUS

acclaiming, 2020-08-08:
Manjaro issues now fixed: the tablet tip and side button binding works absolutely perfectly on linux

[image]

X9VoiD, 2020-08-08:
Support for Windows RawInput just got merged recently, next release will have it.

InfinityGhost, 2020-08-12:
Can we confirm if this is fixed at bcce99be919cb6e50131de437be7d139c880b322?

InfinityGhost, 2020-08-16:
Assuming fixed due to inactivity.

closed 2020-08-16 by InfinityGhost