OpenTabletDriver/OpenTabletDriver #280: Add 32 bit installer for windows

EutMeHardPlz, 2020-09-09:
Please add 32bit installer, my laptop says its for 64bit arch

InfinityGhost, 2020-09-09:
Windows x86 as a whole is completely untested, however we can try building OTD for x86 to see if it works out of the box.

InfinityGhost, 2020-09-11:
Compiling for win-x86 appears to work no problem, but I only tested it on a x64 platform. I don't intend on supporting x86 anyway, but I'll provide build instructions here
# Building for Windows x86 (32-bit)
- Install [git](https://git-scm.com/)
- Install [.NET 5 SDK](https://dotnet.microsoft.com/download/dotnet/5.0)
- Run the following commands in PowerShell
```ps
git clone https://github.com/InfinityGhost/OpenTabletDriver src
mkdir OpenTabletDriver
cd src
Copy-Item -Recurse .\TabletDriverLib\Configurations ..\OpenTabletDriver\Configurations
dotnet publish -f net5 -r win-x86 -o ..\OpenTabletDriver
```
- You can now run `OpenTabletDriver.UX.Wpf.exe` in the OpenTabletDriver directory that was created. It should launch the daemon as well.

closed 2020-09-11 by InfinityGhost

oDruvee, 2020-12-14:
I tried running the commands in the GIT Powershell and when I get to the Copy Item it says Copy Item: command not found

InfinityGhost, 2020-12-14:
Let me see if I can make you up a binary myself

InfinityGhost, 2020-12-14:
[OpenTabletDriver win-x86.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5686758/OpenTabletDriver.win-x86.zip)
This was built on the v0.4.2 tag

oDruvee, 2020-12-14:
Ok, Thanks alot!

oDruvee, 2020-12-14:
How can I get it to bypass my Avast Antivirus?

InfinityGhost, 2020-12-14:
That's something you'll have to figure out with Avast. Most likely an antivirus exclusion.

oDruvee, 2020-12-14:
I'm sorry if I'm giving you trouble but I have a Gaomon S620 an it's not detecting the tablet, what can I do?

InfinityGhost, 2020-12-14:
From the OpenTabletDriver Discord
[image]