OpenTabletDriver/OpenTabletDriver #289: Driver can't connect to Daemon

Nyarogami, 2020-09-12:
## Description
When the driver starts, it say's Connecting to OpenTabletDriver Daemon, but after a few minutes it's just closes.

## System Information:
Operating System  Windows 10 64bit 1809
Software Version    0.3.2
Tablet                     Gaomon s620

AbstractQbit, 2020-09-12:
Try using recent builds from github.com/InfinityGhost/OpenTabletDriver/actions

Mrcubix, 2020-09-12:
What happen is simple, either you started the daemon first and it spammed error and caused the UX to not work, or you just haven't started the daemon first

Nyarogami, 2020-09-13:
I can connect to Daemon on 0.3.0, but still can't connect to Daemon on versions: 0.3.2 and 0.3.1.

Mrcubix, 2020-09-13:
come on the official discord server, we will try to fix a fix to your issue there https://discord.gg/P3GYdUS

Nyarogami, 2020-09-13:
Joined

Mrcubix, 2020-09-14:
need to be closed

InfinityGhost, 2020-09-14:
Issue already resolved in master branch

closed 2020-09-14 by InfinityGhost