OpenTabletDriver/OpenTabletDriver #292: Solid light on tablet but not responsive after update.

LeiteDigital, 2020-09-12:
After the update on 11/09/2020 
The table is detected but has a solid light. Done the basic troubleshooting like moving to a different port. The only thing I can do to get it to work is running the driver script from their website [here](https://www.xp-pen.com/download-56.html) but I can't use relative mode as their software doen't support it. Thanks in advance and making a great app for us!

Operating System: Manjaro Linux
KDE Plasma Version: 5.19.5
KDE Frameworks Version: 5.73.0
Qt Version: 5.15.0
Kernel Version: 5.8.6-1-MANJARO
OS Type: 64-bit
Processors: 16 × AMD Ryzen 7 2700X Eight-Core Processor
Memory: 15,7 GiB of RAM
Graphics Processor: GeForce GTX 1070 Ti/PCIe/SSE2
Software Version  | V0.3.2.r110.g01e068b-1 Installed from Manjaro Package Manager
Tablet            |  XP-Pen G430 S

Mrcubix, 2020-09-12:
if you updated, just know that you probably have to remove your OTD settings in Appdata/local/opentabletdriver a file literally called settings.json

LeiteDigital, 2020-09-12:
> ~if you updated, just know that you probably have to remove your OTD settings in Appdata/local/opentabletdriver a file literally called settings.json~
> 
> a string in json has been changed from "OutputMode": "TabletDriverLib.Output.AbsoluteMode",
> to "OutputMode": "OpenTabletDriver.Output.AbsoluteMode", in the new update, so change your OutputMode to fix your issue
It's no longer a solid light since I was messing with it earlier. I did find that .json file by navigating home>user>.config>OpenTabletDriver . I made changes and restarted the software and yes it did change the settings. Just wanted to clarify that switching modes isn't the issue (it was a reason for wanting to use this software as the driver provided by xp pen does not have relative mode). The tablet is not working at all even though it is detected by the software. Is there a way to revert the update?

Mrcubix, 2020-09-12:
well, just delete settings.json then and redo your settings,

LeiteDigital, 2020-09-12:
> well, just delete settings.json then and redo your settings,
> 
> also, what i was talking about is a change in the OutputMode string in the latest masters
Stopped the software, deleted the settings file and then restarted. Unfortunately, the problem persists.  Also, OutputMode string already had the format you suggested to change.  I did change the modes back and forth to see if it would do anything. Thanks for quick replies. Maybe deleting the whole folder after removing the software might help..

Mrcubix, 2020-09-12:
well join the official discord server, we'll see if we can help you there https://discord.gg/P3GYdUS

InfinityGhost, 2020-09-14:
Any update on this?

LeiteDigital, 2020-09-14:
> Any update on this?
Nothing yet, I have been messing around with it here and there. 
Here is the Discord conversation so far 
[image]

LeiteDigital, 2020-09-14:
It's fixed! Thanks to X9VoiD for helping!
Steps:

Navigate to /usr/share/OpenTabletDriver/Configurations/XP-Pen/
Delete the file by doing this in terminal:
sudo rm /usr/share/OpenTabletDriver/Configurations/XP-Pen/G430S_B.json

Download the config file to your downloads folder Here is the link from discord [Download ](https://discord.com/channels/615607687467761684/723399565780189234/755072082806571129)

Then copy the new config file over by doing (assuming you downloaded it to your downloads folder):
sudo cp /home/Yourusernamehere/Downloads/G430S_B.json /usr/share/OpenTabletDriver/Configurations/XP-Pen/

Run this in terminal:
systemctl --user restart opentabletdriver

Close the Software and Open it again.

All credit to X9VoiD and thanks for the clear instructions!

[image]

[image]

closed 2020-09-14 by LeiteDigital