OpenTabletDriver/OpenTabletDriver #308: XP-Pen Deco 01 V2 not detected

drunkfromcoffee, 2020-09-20:
## Description

My tablet isn't detected by OpenTabletDriver. I suspect it's not from OpenTablet's end, as I heard XP-Pen sometime changes it's product's id or something, but I'd very much like to be able to use this driver, since I also use Arch on another setup.

## System Information:

Option            | Value
------------------|------------------
Operating System  | Windows 10
Software Version  | 0.4.0
Tablet            | XP-Pen Deco 01 V2

https://hastebin.com/uhijoliwox.cs

drunkfromcoffee, 2020-09-20:
Looked at the json I sent, apparently the Product ID is 2309 for me, compared to the 2306 from the config. Even if I configured the proper Product ID though, it still doesn't detect it.

Mrcubix, 2020-09-20:
you might want to join the discord server and #help-windows channel https://discord.gg/P3GYdUS

X9VoiD, 2020-09-21:
Open `Configurations/XP-Pen/Deco 01 v2.json` then change ProductID from 2306 to 2309 as you've discovered. Save then restart daemon and ux

X9VoiD, 2020-09-22:
As Mrcubix have said, join our discord server for further assistance: https://discord.gg/P3GYdUS

InfinityGhost, 2020-10-01:
Is this resolved?

InfinityGhost, 2020-10-11:
This issue seems dead. Assuming it has been resolved since.

closed 2020-10-11 by InfinityGhost