OpenTabletDriver/OpenTabletDriver #309: Cannot detect tablet after running xp-pen-tablet driver.

solonovamax, 2020-09-21:
## Description
### Expected behaviour
When running the [manufacturer provided drivers](https://aur.archlinux.org/packages/xp-pen-tablet/) (the `xp-pen-tablet` binary.), closing them, and then running `otd detect`, the tablet should be detected and enabled.
### Current behaviour
After closing the [manufacturer provided drivers](https://aur.archlinux.org/packages/xp-pen-tablet/), the tablet will not be detected by the driver.
### Steps to reproduce
1. Plug in tablet. (Presumably works with all XP-Pen tablets, though only confirmed on the XP-Pen Star G640. Might/might not also happen for other manufacturers.)
2. Run the `xp-pen-tablet`.
3. Close the manufacturer drivers. (when closed, it leaves no background processes)
4. Start otd.
5. Tablet is not detected by `otd detect`.

Note: unplugging & replugging the tablet fixes this issue.

## System Information:

Option            | Value
------------------|------------------
Operating System  |  Arch Linux 5.8.8-arch1-1
Software Version  |  0.4.0-r110.g01e068b
Manufacturer Software Version | 1.2.13
Tablet            |  XP-Pen Star 640

X9VoiD, 2020-09-21:
Any other drivers shouldn't be used when running OTD. Uninstall them.

solonovamax, 2020-09-22:
Yes, but I'm not using it *while* I'm using OTD.
Neither of the programs are running at the same time.

X9VoiD, 2020-09-22:
The drivers configure the tablet in such a way that OTD won't be able to access the tablet, even after killing the driver.

solonovamax, 2020-09-22:
Well, that's a shame. Guess I'll just have to deal with how it is.

closed 2020-09-22 by solonovamax

Estebiu, 2022-06-09:
Hey, I'm sorry to necropost but, how did you uninstall the old xp-pen-tablet drivers? I'm also using arch and a xp640, but downloaded the aur package for the official drivers. How do I remove them? I did remove the package but nothing changed..

gonX, 2022-06-09:
@Estebiu when you installed the AUR package there should have been on-screen instructions telling you to blacklist the driver.

Follow the "Device is currently in use" instructions on the [Linux FAQ](https://opentabletdriver.net/Wiki/FAQ/Linux#argumentoutofrangeexception)

Estebiu, 2022-06-09:
Thank you! That fixed it for me.