OpenTabletDriver/OpenTabletDriver #310: Huion H430P - not detected

LittleWhole, 2020-09-21:
## Description

The Huion H430P tablet is not detected by OpenTabletDriver. It detects my XP-Pen G430S just fine.

I saw that in the Projects > Tablet Support, The Huion H430P is listed as "untested". I've used the XP-Pen G430S just fine with OpenTabletDriver, however my Huion H430P is not being detected by OpenTabletDriver no matter what I do. I can confirm it's not a tablet issue as it works fine with other devices.

## System Information:

Option            | Value
------------------|------------------
Operating System  | Pop!_OS 20.04 LTS
Software Version  | v0.4.0
Tablet            | Huion H430P

X9VoiD, 2020-09-21:
Open `Configurations/Huion/H430P.json` then change ProductID from 110 to 109, then restart daemon and ux

LittleWhole, 2020-09-21:
Sadly doesn't seem to be working

X9VoiD, 2020-09-21:
Join Discord server so we could help you: https://discord.gg/P3GYdUS

closed 2020-09-22 by LittleWhole