OpenTabletDriver/OpenTabletDriver #311: Simpler selection of which monitor the tablet will map to.

solonovamax, 2020-09-22:
## Description
Allow users to select monitors rather than having to fine-tune the display area settings.

## Example implementation:
For me, I have 3 monitors and finding the correct X Offset took a bit of fiddling because it seems that the offsets aren't relative to the top right corner.
Instead, it would show a dropdown box with where you can select "Custom" or "Monitor". Under "Custom", it will show the configuration for the display area. Under "Monitor", it will show a dropdown box with a list of all your screens (and an additional "all screens") + a display preview for the one you selected. (see attached for example.)
There would also be lines drawn from an example image of a tablet onto the screen to show how it would be mapped to the screen.

### Single monitor
[image]
### All monitors
[image]
### Full window
[image]
(why did I spend 3 hours making this)

#### Notes
- The drop-down menus are all expanded only to show you what would be in each one.
- In my specific configuration I have a 1680×1050 monitor as my left-most monitor, the rest are 1080p.
- The screen names here are not tied directly to the X monitor names, they are sorted based on their position.
- The coordinates shown in the picture are the top-left coordinates to the bottom right coordinates of the X monitor
- Even though my 1st monitor is 1680×1050, I still stretched the screen size in the display, as it is roughly the same physical size. (Where the pen appears should be proportional to physical monitor size rather than display size as that makes more sense when using it. Though, then x coordinate on the tablet will already be more or less multiplied by 3, so it isn't exactly the easiest to use.)

X9VoiD, 2020-09-22:
Right clicking on display area will provide you an option to set to specific monitors

solonovamax, 2020-09-23:
Oh, I'm just braindead.

Would be nice to have a preview for it though because  f a n c y

solonovamax, 2020-09-23:
Also, I'd like to add on that a keybind to cycle through which monitor its bound to might be nice for working with multimonitor setups. (rather than having to go to your mouse for a simple task, you can use the tablet)
Alternatively, you could add a keybind to go left/right in your monitor selection.

(Don't feel like making a new issue for this since these are kind of related.)

X9VoiD, 2020-09-24:
This kind of functionality is not essential, nor in high demand, so if ever this feature will be implemented, it will be as a plugin. Not as a core part of OTD. The issue may be closed now

solonovamax, 2020-09-25:
Fair

closed 2020-09-25 by solonovamax