OpenTabletDriver/OpenTabletDriver #312: Make display area and tablet area use the top left corner for offsets

solonovamax, 2020-09-22:
## Description
With the current implementation, the X Offset and Y Offset properties of the "Display Area" and "Tablet Area" sections indicate the offset of the *center* of the selection.
I thought this was rather confusing while trying to use the program and think it might be more intuitive if the X Offset and Y Offset properties indicated the offset of the top left corner. So, if you wanted your tablet positioned in the top left corner, then the offset would be 0,0. That way if you resized it would also remain in the top right corner. Additionally, the offset would be 0,0 for a tablet that uses the full area.

This kind of became an issue for me while trying to select the correct display area, because I have 3 screens and I wanted to put it on the middle one. Since there was no explanation I tried many things and eventually figured it out through trial and error. (And later realized that it was the offset for the middle of the area and not the top left corner.)

If you choose not to make it relative to the top right corner, at least add some form of tool tip that lets people know it is relative to the center of the area.

InfinityGhost, 2020-09-23:
This functionality will not be changed, however we will draw a marker of some sort on the center of the area to show that its the origin of the area.

solonovamax, 2020-09-23:
That works too. 
Also, as #304 is adding tooltips, you might want to add a tooltip that says that it's relative to the center of the area.

closed 2020-09-26 by InfinityGhost