OpenTabletDriver/OpenTabletDriver #326: Nothing happens when clicking "start" in the updater window.

captainshrimps, 2020-09-27:
## Description

I installed the driver using the windows auto-updater and it installs successfully but when I click start all that happens is the updater window disappears for a moment then re-opens. Nothing else happens and I can't use my tablet. 
[image]

## System Information:

Option            | Value
------------------|------------------
Operating System  | Windows 10
Software Version  | 0.4.0
Tablet            | XP-PEN G430S

X9VoiD, 2020-09-27:
Install .NET Core runtime

captainshrimps, 2020-09-27:
> 
> 
> Install .NET Core runtime

Thank you, this solved the issue.

closed 2020-09-27 by captainshrimps