OpenTabletDriver/OpenTabletDriver #332: Gaomon m106k support?

OneBadPeanut, 2020-09-29:
## Description
 Id like this tablet to be supported so I could play osu with better drivers

InfinityGhost, 2020-09-29:
Export diagnostics in the GUI and send it here. That can help us get a base to start off with.

OneBadPeanut, 2020-09-29:
How would I do that?

InfinityGhost, 2020-09-29:
- Start `OpenTabletDriver.UX.Wpf.exe` (assuming you're on Windows)
- In the menu, click Help -> Export diagnostics
- This will show a save file dialog, which you can just save it and drag it here or upload and link

OneBadPeanut, 2020-09-29:
Alright this may be it. Sorry if it is a bit cluttered i wasnt able to make a link or zip
{
  "App Version": "OpenTabletDriver v0.4.0",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "6.2.9200.0",
    "VersionString": "Microsoft Windows NT 6.2.9200.0"
  },
  "Environment Variables": {
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "USERNAME": "AJ",
    "ComSpec": "C:\\WINDOWS\\system32\\cmd.exe",
    "APPDATA": "C:\\Users\\AJ\\AppData\\Roaming",
    "ProgramFiles": "C:\\Program Files",
    "CUDA_PATH_V11_0": "C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v11.0",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "TMP": "C:\\Users\\AJ\\AppData\\Local\\Temp",
    "HOMEPATH": "\\Users\\AJ",
    "TEMP": "C:\\Users\\AJ\\AppData\\Local\\Temp",
    "SESSIONNAME": "Console",
    "CUDA_PATH": "C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v11.0",
    "windir": "C:\\WINDOWS",
    "PUBLIC": "C:\\Users\\Public",
    "SystemDrive": "C:",
    "NVCUDASAMPLES_ROOT": "C:\\ProgramData\\NVIDIA Corporation\\CUDA Samples\\v11.0",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "PROCESSOR_LEVEL": "23",
    "OneDriveConsumer": "C:\\Users\\AJ\\OneDrive",
    "PROCESSOR_REVISION": "0802",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "COMPUTERNAME": "AJM-PC",
    "NVCUDASAMPLES11_0_ROOT": "C:\\ProgramData\\NVIDIA Corporation\\CUDA Samples\\v11.0",
    "LOGONSERVER": "\\\\AJM-PC",
    "HOMEDRIVE": "C:",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "USERDOMAIN_ROAMINGPROFILE": "AJM-PC",
    "FPS_BROWSER_USER_PROFILE_STRING": "Default",
    "SystemRoot": "C:\\WINDOWS",
    "USERPROFILE": "C:\\Users\\AJ",
    "USERDOMAIN": "AJM-PC",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "NUMBER_OF_PROCESSORS": "12",
    "OS": "Windows_NT",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "FPS_BROWSER_APP_PROFILE_STRING": "Internet Explorer",
    "Path": "C:\\PROGRAM FILES\\NVIDIA GPU COMPUTING TOOLKIT\\CUDA\\V11.0\\BIN;C:\\PROGRAM FILES\\NVIDIA GPU COMPUTING TOOLKIT\\CUDA\\V11.0\\LIBNVVP;C:\\PROGRAM FILES (X86)\\COMMON FILES\\ORACLE\\JAVA\\JAVAPATH;C:\\WINDOWS\\SYSTEM32;C:\\WINDOWS;C:\\WINDOWS\\SYSTEM32\\WBEM;C:\\WINDOWS\\SYSTEM32\\WINDOWSPOWERSHELL\\V1.0\\;C:\\WINDOWS\\SYSTEM32\\OPENSSH\\;C:\\PROGRAM FILES (X86)\\NVIDIA CORPORATION\\PHYSX\\COMMON;;C:\\PROGRAM FILES\\NVIDIA CORPORATION\\NVIDIA NVDLISR;;C:\\PROGRAM FILES\\NVIDIA CORPORATION\\NSIGHT COMPUTE 2020.1.2\\;C:\\Users\\AJ\\AppData\\Local\\Microsoft\\WindowsApps;",
    "OneDrive": "C:\\Users\\AJ\\OneDrive",
    "LOCALAPPDATA": "C:\\Users\\AJ\\AppData\\Local",
    "NVTOOLSEXT_PATH": "C:\\Program Files\\NVIDIA Corporation\\NvToolsExt\\",
    "ProgramData": "C:\\ProgramData",
    "PROCESSOR_IDENTIFIER": "AMD64 Family 23 Model 8 Stepping 2, AuthenticAMD",
    "ProgramW6432": "C:\\Program Files"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#hid&col04#1&2d595ca7&5&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 255,
      "ProductID": 47820,
      "ReleaseNumberBcd": 1,
      "ReleaseNumber": "0.1",
      "ProductVersion": 1,
      "Manufacturer": "DJP Inc.",
      "ProductName": "Virtual Multitouch Device",
      "SerialNumber": "123123123",
      "MaxInputReportLength": 8,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
[... 1364 lines cut ...]
      "Time": "2020-09-29T13:17:53.737975-04:00",
      "Group": "Settings",
      "Message": "Tip Binding: 'Mouse Button Binding: Left'@1%",
      "Level": 1
    },
    {
      "Time": "2020-09-29T13:17:53.7379796-04:00",
      "Group": "Settings",
      "Message": "Pen Bindings: [0, ], [1, ]",
      "Level": 1
    },
    {
      "Time": "2020-09-29T13:17:53.7379843-04:00",
      "Group": "Settings",
      "Message": "Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]",
      "Level": 1
    },
    {
      "Time": "2020-09-29T13:17:53.7379889-04:00",
      "Group": "Settings",
      "Message": "Driver is auto-enabled.",
      "Level": 1
    }
  ]
}

OneBadPeanut, 2020-09-29:
Alright this should be it this time. Better than my last post which was a bigline of text
[gaomon m106k.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5300826/gaomon.m106k.txt)

InfinityGhost, 2020-09-29:
Writing this here for reference
```json
{
  "VendorID": 9580,
  "ProductID": 110,
  "MaxInputReportLength": 8
}
```

InfinityGhost, 2020-09-29:
Does your release have `OpenTabletDriver.Console.exe` in it?

OneBadPeanut, 2020-09-29:
yes it does

InfinityGhost, 2020-09-29:
Save this text as `Configurations/Gaomon/M106K.json` in the configurations folder
```json
{
  "Name": "Gaomon M106K",
  "DigitizerIdentifier": {
    "VendorID": 9580,
    "ProductID": 110,
    "InputReportLength": 8,
    "OutputReportLength": 0,
    "ReportParser": "OpenTabletDriver.Vendors.Gaomon.GaomonReportParser",
    "FeatureInitReport": null,
    "OutputInitReport": null
  },
  "AlternateDigitizerIdentifier": {
    "VendorID": 0,
    "ProductID": 0,
    "InputReportLength": 0,
    "OutputReportLength": 0,
    "ReportParser": null,
    "FeatureInitReport": null,
    "OutputInitReport": null
  },
  "AuxilaryDeviceIdentifier": {
    "VendorID": 0,
    "ProductID": 0,
    "InputReportLength": 0,
    "OutputReportLength": 0,
    "ReportParser": null,
    "FeatureInitReport": null,
    "OutputInitReport": null
  },
  "Width": 254.0,
  "Height": 158.8,
  "MaxX": 50800.0,
  "MaxY": 31760.0,
  "MaxPressure": 2047,
  "ActiveReportID": null,
  "Attributes": {},
  "DeviceStrings": {},
  "InitializationStrings": []
}
```
We still need to get device strings from the tablet with the console app, however.

InfinityGhost, 2020-09-29:
- Make sure the GUI is closed
- Using powershell, `cd` into the directory with `OpenTabletDriver.Console.exe`
- From there run the following, which if successful will return a string and save it to `201.txt` which we will use for identification.
```ps
./OpenTabletDriver.Console.exe getstring 201 | Out-File ./201.txt
```
- Send that file or its contents here.

OneBadPeanut, 2020-09-29:
I think I might've done something wrong but I will send anyways
[201.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5301013/201.txt)

InfinityGhost, 2020-09-29:
oops, forgot we need a development build for the getstring command. https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/278801343

InfinityGhost, 2020-09-29:
You'll have to close the daemon and recopy that configuration, but otherwise redo the previous steps with the development build.

OneBadPeanut, 2020-09-29:
[201.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5301056/201.txt)
I got this once I did that

InfinityGhost, 2020-09-29:
This configuration should work *almost* correctly now, however we still need one more thing, the active report range. This is where you use the tablet debugger to see what the ReportID property changes to when dragging, hovering, and when the pen is off of the tablet.
```json
{
  "Name": "Gaomon M106K",
  "DigitizerIdentifier": {
    "VendorID": 9580,
    "ProductID": 110,
    "InputReportLength": 8,
    "OutputReportLength": 0,
    "ReportParser": "OpenTabletDriver.Vendors.Gaomon.GaomonReportParser",
    "FeatureInitReport": null,
    "OutputInitReport": null
  },
  "AlternateDigitizerIdentifier": {
    "VendorID": 0,
    "ProductID": 0,
    "InputReportLength": 0,
    "OutputReportLength": 0,
    "ReportParser": null,
    "FeatureInitReport": null,
    "OutputInitReport": null
  },
  "AuxilaryDeviceIdentifier": {
    "VendorID": 0,
    "ProductID": 0,
    "InputReportLength": 0,
    "OutputReportLength": 0,
    "ReportParser": null,
    "FeatureInitReport": null,
    "OutputInitReport": null
  },
  "Width": 254.0,
  "Height": 158.8,
  "MaxX": 50800.0,
  "MaxY": 31760.0,
  "MaxPressure": 2047,
  "ActiveReportID": null,
  "Attributes": {},
  "DeviceStrings": {
    "201": "OEM02_T151_\\d{6}$"
  },
  "InitializationStrings": [
    200
  ]
}
```

InfinityGhost, 2020-09-29:
> I got this once I did that, was daemon supposed to be closed?

If I understand what you meant, the daemon exited after requesting the string? That's a separate issue on its own I'll tackle another time.

OneBadPeanut, 2020-09-29:
It says no tablet detected so should i still use the dev build or switch to the other one

InfinityGhost, 2020-09-29:
Does it still not detect if you change the DeviceStrings property to empty?
```json
"DeviceStrings": {},
```

OneBadPeanut, 2020-09-29:
it still is not detected. F

InfinityGhost, 2020-09-29:
Re-export diagnostics and resend. I think something might have changed.

OneBadPeanut, 2020-09-29:
[gaomon m106k 2.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5301307/gaomon.m106k.2.txt)
this is it now i left in everything this time so hopefully it helps

InfinityGhost, 2020-09-29:
Okay, try setting this property to empty, restart the daemon and gui
```json
  "InitializationStrings": []
```

OneBadPeanut, 2020-09-29:
no but i got this as a message 
[image]

InfinityGhost, 2020-09-29:
This is gonna require USB packet sniffing the Gaomon driver to figure out what they're doing to open the device stream. https://discord.gg/P3GYdUS and later on we'll see what we can do.

OneBadPeanut, 2020-09-29:
alright then I appreciate the help though

OneBadPeanut, 2020-09-29:
This is what I also get in daemon.exe
[image]

closed 2020-09-30 by InfinityGhost