OpenTabletDriver/OpenTabletDriver #334: No tablet detected with HUION HS64

zMici, 2020-09-29:
## Description

The drivers aren't detecting my tablet,

## System Information:

Option            | Value
------------------|------------------
Operating System  |    
[image]
Windows 10
Software Version  |    0.4.0
Tablet            |    Huion HS64

Mrcubix, 2020-09-29:
make sure all other driver are uninstalled like vmulti, huion driver, also, uninstall winusb in programs if not already done, finally, go into device manager, delete "huion tablet" and "Graphic Tablet Device", reboot and it should be detected,

zMici, 2020-09-29:
Thank you, I guess I had vmulti still installed. works perfectly now.

closed 2020-09-29 by zMici