OpenTabletDriver/OpenTabletDriver #336: [Windows] Wrong input mapping when monitors are unaligned

X9VoiD, 2020-09-30:
## Description
When the monitors are not aligned in a specific axis, OTD's input behaves incorrectly. It tries to scale from the lowest value of that axis to the highest without regards to where the monitors end in that axis. 

## System Information:
Option            | Value
------------------|------------------
Operating System  |  Windows

X9VoiD, 2020-09-30:
This is an image for example.

[image]

The cursor won't be able to reach the bottom part of the mapped monitor area by the same scaled amount of misalignment in Y axis

InfinityGhost, 2020-10-03:
Unable to reproduce this as of right now

X9VoiD, 2020-10-05:
Closing this issue

closed 2020-10-05 by X9VoiD