OpenTabletDriver/OpenTabletDriver #341: How to make this work on Gaomon S620

UltimateKaren, 2020-10-02:
Hello! I know this isn't an issue but I couldn't find anything in the FAQ that helped me. I use a gaomon s620 and downloaded and extracted everything in OpenTabletDriver.win-x64. I ran OpenTabletDriver.Daemon.exe and OpenTabletDriver.UX.Wpf.exe and it hasn't detected my tablet and isn't showing any settings. Here are some screenshots
I'm bad with computers in general and if someone could help out that would be greatly appreciated :"D

[image]
[image]

X9VoiD, 2020-10-02:
1. Open Device Manager
2. View -> Devices by Connection
3. Find Gaomon HID / Graphics Tablet Device in USB controllers
4. Right Click -> Uninstall Device (tick the checkbox that says "Delete driver files")
5. Replug tablet
6. Open OTD

If still not working, join the discord server: https://discord.gg/P3GYdUS

InfinityGhost, 2020-10-07:
Issue appears to be stale, no update after 1 week and this will be closed.

closed 2020-10-11 by InfinityGhost

bossdragonde, 2020-10-14:
Thanks this fixed my problem.

InfinityGhost, 2020-10-14:
Added this to the [Windows FAQ](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Windows-FAQ#my-gaomonhuion-tablet-isnt-detected-but-is-listed-as-supported)