OpenTabletDriver/OpenTabletDriver #347: Daemon stops responding input when Task Manager is focused

aticie, 2020-10-03:
Here's a quick demonstration of the bug. I'm controlling the cursor with tablet throughout the video until very end.

https://streamable.com/07d80b

To reproduce:

1. Open up daemon along with task manager. 
1. Observe that it is not possible to click on Task Manager window with tablet.
1. Click on Task Manager icon from taskbar.
1. Tablet input gets blocked.

Tablet: Wacom Ctl-480

Console Logs:

[Detect:Info]	The plugin directory 'C:\Users\Efehan\AppData\Local\OpenTabletDriver\Plugins' has been created
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Huion H430P'
[Detect:Info]	Searching for tablet 'Huion H610 Pro'
[Detect:Info]	Searching for tablet 'Huion H640P'
[Detect:Info]	Searching for tablet 'Huion HS610'
[Detect:Info]	Searching for tablet 'Huion HS611'
[Detect:Info]	Searching for tablet 'Huion HS64'
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom CTH-470'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[Detect:Info]	Found device 'Intuos PS'.
[Detect:Info]	Using report parser type 'OpenTabletDriver.Tablet.TabletReportParser'.
[Detect:Debug]	Device path: \\?\hid#vid_056a&pid_030e&mi_00#8&18435306&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
[Detect:Debug]	Setting feature: 02-02
[Detect:Debug]	Found aux device with report length 64.
[Detect:Debug]	Device path: \\?\hid#vid_056a&pid_030e&mi_01#8&2b736f44&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
[IPC:Debug]	Connected to a client.
[Settings:Info]	Display area: [1920x1080@<960, 540>:0°],
[Settings:Info]	Output mode: OpenTabletDriver.Output.AbsoluteMode
[Settings:Info]	Tablet area: [152x95@<76, 47.5>:0°],
[Display:Debug]	DPI Awareness enabled
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Tip Binding: 'Mouse Button Binding: Left'@1%
[Settings:Info]	Pen Bindings: [0, ], [1, ]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info]	Driver is auto-enabled.
[Settings:Info]	Output mode: OpenTabletDriver.Output.AbsoluteMode
[Settings:Info]	Display area: [1920x1080@<960, 540>:0°],
[Settings:Info]	Tablet area: [80x45@<46.972122, 39.913204>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Tip Binding: 'Mouse Button Binding: Left'@1%
[Settings:Info]	Pen Bindings: [0, ], [1, ]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info]	Driver is auto-enabled.
[Settings:Info]	Output mode: OpenTabletDriver.Output.AbsoluteMode
[Settings:Info]	Display area: [1920x1080@<960, 540>:0°],
[Settings:Info]	Tablet area: [80x45@<46.972122, 39.913204>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Tip Binding: 'Mouse Button Binding: Left'@1%
[Settings:Info]	Pen Bindings: [0, ], [1, ]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info]	Driver is auto-enabled.
[Settings:Info]	Output mode: OpenTabletDriver.Output.AbsoluteMode
[Settings:Info]	Display area: [1920x1080@<960, 540>:0°],
[Settings:Info]	Tablet area: [80x45@<46.972122, 39.913204>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Tip Binding: 'Mouse Button Binding: Left'@1%
[Settings:Info]	Pen Bindings: [0, Mouse Button Binding: Middle], [1, Mouse Button Binding: Right]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info]	Driver is auto-enabled.
[Settings:Info]	Output mode: OpenTabletDriver.Output.AbsoluteMode
[Settings:Info]	Display area: [1920x1080@<960, 540>:0°],
[Settings:Info]	Tablet area: [80x45@<46.972122, 39.913204>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Tip Binding: 'Mouse Button Binding: Left'@1%
[Settings:Info]	Pen Bindings: [0, Mouse Button Binding: Middle], [1, Mouse Button Binding: Right]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info]	Driver is auto-enabled.
[Settings:Info]	Output mode: OpenTabletDriver.Output.AbsoluteMode
[Settings:Info]	Display area: [1920x1080@<960, 540>:0°],
[Settings:Info]	Tablet area: [80x45@<46.972122, 39.913204>:0°],
[Settings:Info]	Clipping: Enabled
[Settings:Info]	Tip Binding: 'Mouse Button Binding: Left'@1%
[Settings:Info]	Pen Bindings: [0, Mouse Button Binding: Right], [1, Mouse Button Binding: Middle]
[Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ]
[Settings:Info]	Driver is auto-enabled.

X9VoiD, 2020-10-03:
This behaviour is not a bug, and is a Windows limitation for security reasons. If you want a "fix" for that, use the VMultiMode plugin from [VoiDPlugins](https://github.com/X9VoiD/VoiDPlugins)

X9VoiD, 2020-10-03:
This issue may be closed now since it's not really an issue (more of a limitation of doing everything in usermode) and can be mitigated with a plugin

closed 2020-10-03 by InfinityGhost