OpenTabletDriver/OpenTabletDriver #350: NEW 1060PLUS (2048) not detected

gowcoizer, 2020-10-04:
## Description

"NEW 1060PLUS" has some model with 2048 pressure sensitivity, this what I got. I tested this software cause of troubles with the original drivers and this software doesn't detect the "NEW 1060PLUS" plus
here are all the 1060 model variations and drivers. I will check from time to time if this gets support for what I got. Keep up
 

## System Information:

Windows 10 Enterprise     | Versoin 2004 (OS BUILD 19041.508
Tablet            |  NEW 1060PLUS (2048)

X9VoiD, 2020-10-04:
New 1060 Plus is already tested to be working.

If it's still not detected on v0.4.1 then do the following:

1. Open Device Manager
2. Find Graphics Hid Tablet
3. Right Click -> Uninstall (tick the "Delete driver file" checkbox)
4. Replug tablet
5. Restart OTD

InfinityGhost, 2020-10-04:
> New 1060 Plus is already tested to be working.
> 
> If it's still not detected on v0.4.1 then do the following:
> 
>     1. Open Device Manager
> 
>     2. Find "Graphics Tablet Device"
> 
>     3. Right Click -> Uninstall (tick the "Delete driver file" checkbox)
> 
>     4. Replug tablet
> 
>     5. Restart OTD

2048 pressure level model might have different identifiers

InfinityGhost, 2020-10-04:
Export diagnostic information in the GUI (Help -> Export Diagnostics) and send here.

gowcoizer, 2020-10-05:
NEW 1060 Plus (2048) Diagnostics

[https://pastebin.com/sX9UrdSd](url)

aidanw, 2020-10-12:
Just for extra info my 1060 plus (that I presume must be the 2048 version, though it does not say on it!) generates this info in your log file.

This is with no Huion drivers loaded, and your software trying to detect it (obviously!)

    {
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006e&mi_01&col01#8&3a3b28e0&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 9580,
      "ProductID": 110,
      "ReleaseNumberBcd": 12288,
      "ReleaseNumber": "30.0",
      "ProductVersion": 12288,
      "Manufacturer": "",
      "ProductName": "PenTablet ",
      "SerialNumber": "",
      "MaxInputReportLength": 8,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 2
    },

Hydra82, 2020-10-20:
My Huion 1060 Plus also isn't working heres the logfile for the tablet
```

{
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006e&mi_00#8&346e00af&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 9580,
      "ProductID": 110,
      "ReleaseNumberBcd": 12288,
      "ReleaseNumber": "30.0",
      "ProductVersion": 12288,
      "Manufacturer": "",
      "ProductName": "",
      "SerialNumber": "",
      "MaxInputReportLength": 12,
      "MaxOutputReportLength": 0,
      "MaxFeatureReportLength": 0
    },
```

InfinityGhost, 2020-10-20:
For anyone who is interested in fixing this, work is being done in #318 and at this point we just need a USB packet capture (with [usbpcap](https://desowin.org/usbpcap/)) with the XP-Pen driver to get the initialization device strings

gowcoizer, 2020-10-20:
I got notifications but not experienced with github and it's logic, so is
this something concerning me¿ I mean I want testing with usbcap and XP-pen
drivers? And report ¿ Do I mean Huion drivers? I want to contribute

InfinityGhost, 2020-10-20:
I meant the Huion driver lol

InfinityGhost, 2020-11-05:
Any update on this?

gowcoizer, 2020-11-07:
Hey there, well I got lots of work to do lately so haven't invested time,
but also newbie to those usbcap stuff so it might be something really
simple but I just wait to have time window to invest time what and how to
do.. if it is just a few clicks just confirm so I will be braver and deliver

Hydra82, 2020-11-08:
if anyone has a step by step guide on how to do the USB packet capture Ill do it

X9VoiD, 2020-11-08:
Download and install [USBPcap](https://desowin.org/usbpcap/).

Also install the official oem driver for New 1060 Plus. Change area to something really small (so we know if driver works) then restart PC.

Once done, double click C:\Program Files\USBPcap\USBPcapCMD.exe

Find the corresponding usbpcap filter that the HID-compliant pen is under. Type that number then press enter.

Type any filename you want with .pcap extension, then press enter.

Move pen for 5 secs.
Reconnect tablet.
Move pen until your tablet area becomes small and then move for 5 secs again.

Press Q to stop the capture. Upload the generated .pcap file here.

gowcoizer, 2020-11-08:
Dunno if did it correct but may be.. what I did is started the capturing applying the filtering, moved pen around for some time, disconnected... then connected and moved again... than decreased the area size via theoriginal driver a lot... moved again on the reduced area... than disconnected and connected again and moved some more... let me know if this is enough 

[HUION_1060_(2048).zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5507070/HUION_1060_.2048.zip)

X9VoiD, 2020-11-08:
It wasn't done correctly, it's likely that you chose the wrong filter.

X9VoiD, 2020-11-08:
If you're not sure of which to choose, you may send output of USBPcap here during filter selection

gowcoizer, 2020-11-08:
By filter do you mean port number, I wrote just a number.. didn't quoted whle name.. for example "Port 1".. I've done it lame probably... here's the list... help and thank you
[image]

X9VoiD, 2020-11-08:
Type
```
3
```

gowcoizer, 2020-11-08:
Now I see.. file is bigger so more info captured... uploading.. here it is... so describing what I did while capturing...
moved cursor with pen for a few seconds, then disconnected than connected again moved a little more, reduced area, moved again for a few seconds and Q capturing... let me know if you need something specific
[HUION_1060_(2048).zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5507329/HUION_1060_.2048.zip)

X9VoiD, 2020-11-09:
Download and install this [dev build](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/348464123).

Replace `Configuration/Huion/New 1060 Plus.json` with this one: [New 1060 Plus.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5507419/New.1060.Plus.zip).

Launch OTD then check if it got detected. Export Diagnostics if it did not.

gowcoizer, 2020-11-09:
I did not uninstall the original driver, just closed the GUI.
The downloaded dev build don't seemks installable but portable so started from "OpenTabletDriver.UX.Wpf.exe"
The tablet don't show as detected, I can move the pen tool but there is ghost input from time to time, probably using some native Windows 10 driver.. If I am doing it wrong let me know.
Here is diagnostic:
[Huion_1060_(2048)_diagnostic.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5508930/Huion_1060_.2048._diagnostic.zip)

X9VoiD, 2020-11-09:
Open Task Manager and make sure that there is no other OTD Daemon left running before launching the OTD dev build.

Changed line endings: [New 1060 Plus.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5509361/New.1060.Plus.zip)

gowcoizer, 2020-11-09:
I suppose here must show detected right? Checked with Task manager and  Process Hacker, no open OTD or Opentablet Daemon found
[image]

Not detected, here is diagnostic if useful
[Huion_1060_(2048)_diagnostic_1.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5509998/Huion_1060_.2048._diagnostic_1.zip)

X9VoiD, 2020-11-10:
I'm not sure what it is but just to be sure... close all OTD instances, redownload the [dev build](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/348464123), extract to any non-admin folder, then modify `Configurations\Huion\New 1060 Plus.json` to the following:

```js
{
  "Name": "Huion New 1060 Plus (Test)",
  "DigitizerIdentifiers": [
    {
      "Width": 254.0,
      "Height": 158.75,
      "MaxX": 50800.0,
      "MaxY": 31750.0,
      "MaxPressure": 8191,
      "ActiveReportID": {
        "Start": 64,
        "StartInclusive": true,
        "End": null,
        "EndInclusive": false
      },
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 8,
      "OutputReportLength": null,
      "ReportParser": null,
      "FeatureInitReport": null,
      "OutputInitReport": null,
      "DeviceStrings": {
        "201": "HUION_T174_\\d{6}$"
      },
      "InitializationStrings": [
        200
      ]
    }
  ],
  "AuxilaryDeviceIdentifiers": [],
  "Attributes": {
    "libinputoverride": "1",
    "WinInterface": "01",
    "WinUsage": "03"
  }
}
```
Save the modified file then launch OTD.

X9VoiD, 2020-11-13:
Any news on this?

gowcoizer, 2020-11-13:
Sorry, busy days... tomorrow evening wil supply report. Thanks for trying
to develop, hope to help what I can

person4268, 2020-12-03:
I just tried out your fix, it didn't detect the tablet.
[image]

Here's what I did:
1. Uninstalled original Huion driver
2. Downloaded the linked dev build
3. Replaced the JSON file with the contents in your last post

I've also tried running it as admin with no success.

X9VoiD, 2020-12-03:
Wait until HidSharpCore's  [#14](https://github.com/InfinityGhost/HIDSharpCore/pull/14) reaches OTD

Mrcubix, 2020-12-17:
you can now try the latest master Action build with thta config you're using rn https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/427325474

X9VoiD, 2021-01-03:
@gowcoizer Download this [dev build](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/457171437), delete `%localappdata%/OpenTabletDriver`, and then upload another diagnostics using this build.

X9VoiD, 2021-01-22:
@gowcoizer try out [0.5.0](https://github.com/InfinityGhost/OpenTabletDriver/releases/tag/v0.5.0) release. Export another diagnostics file using the new release.

gowcoizer, 2021-02-02:
Hey there, this is report from v0.5.1 Will this diagnostic help? I find this confusing. I have my original driver installed, when testing just closing my Huion App, and Open driver says "No tablet detected". The tablet worx but probably because of Windows Ink or its original drivers. If you have instructions let me know.

[HUION_1060_(2048) Open Tablet v0.5.1 Diagnostic export.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5913034/HUION_1060_.2048.Open.Tablet.v0.5.1.Diagnostic.export.zip)

Mrcubix, 2021-02-02:
uninstall winUSB then huion drivers in your programs, if WinUSB is still installed and huion drivers are uninstall, re-download and then uninstall them in that order

Razer Synapse is known to cause issues with OTD

Device Manager -> View -> Devices by Container
Find for any of these devices:
  -  Gaomon HID/Tablet
  -  Huion HID/Tablet
  -  Graphics Tablet Device
  -  HID Tablet
Expand drop down then right click on matching device
Click Update -> "Browse my computer" -> "Let me pick" -> HID-compliant device or USB Input Device

once that's done, go back to OTD : help > export diagnostic

gowcoizer, 2021-02-02:
Uninstalled huion driver, What is winUSB, I have no Razer Synapse and 

https://i.imgur.com/vcbLnZv.png

Mrcubix, 2021-02-02:
Same location as huion driver, except you're supposed to uninstall winUSB first and then huion driver, so now you have to reinstall huion driver to uninstall winUSB first and then huion driver

gowcoizer, 2021-02-02:
I really dunno what you are talking about.  Attached is complete filelist in **C:\PenTabletDriver** (this is the original Huion driver installation folder for the Huion drivers for **1060PLUS(2048)**

[C_PenTabletDriver.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5913880/C_PenTabletDriver.txt)

Mrcubix, 2021-02-02:
same location as huion driver aKa "Change or uninstall a program"

gowcoizer, 2021-02-03:
Removed winUSB from Windows 10's Apps and Features. After that removed the Huion driver from its uninstall.exe in its folder. After that opened OpenTabletDriver and tried menus Tablets>Detect Tablet and still no tablet detected below. Here is exported diagnostic
[HUION_1060_(2048) Open Tablet v0.5.1 Diagnostic export.json.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5920224/HUION_1060_.2048.Open.Tablet.v0.5.1.Diagnostic.export.json.zip)

Mrcubix, 2021-02-03:
check again if the devices are removed but i wouldn't get any hope right now since i've seen the content of the diagnostic

gowcoizer, 2021-02-03:
I am lost in those explanations. Seems we are going back and forth. Posted in previous posts image of the devices in device manager, can not locate such name as you suggest. My device isn't removed from the USB port while I test OpenTabletDriver, disconnecting and connecting it while OpenTabletDriver is started to see indication of detecting behaviour, no such. Whatever, let it be as it is. This device is not working for now with the open driver, I am not skilled enough to point why. Thank you for the efforts

Mrcubix, 2021-02-03:
From what i can tell, this tablet won't work on any other driver than Huion's on windows, on linux and mac however, they will, for this tablet to work on windows, [USB.NET](https://github.com/InfinityGhost/USB.NET) is required, which mean you'll have to either wait or switch platform as it's still in development at the moment

Mrcubix, 2021-09-09:
Join the [discord server](https://discord.gg/P3GYdUS) when you have time, WinUSB support has been added in pre-release and it will be faster to get it compatible if you join

jamesbt365, 2021-12-29:
WinUSB support has been added and therefore this config should be detected with a new config added in #1648

closed 2021-12-29 by jamesbt365