OpenTabletDriver/OpenTabletDriver #351: Mouse not moving even though tablet is properly detected. No error logs.

solonovamax, 2020-10-05:
## Description
OTD detects the tablet properly, but then the mouse does not move when using the tablet.
This is probably™ because I upgraded my kernel and forgot to reboot (smart boi), but it'd be nice if it could detect this and throw errors in the log when this happens, rather than silently suppressing it.
(I'll update this later when I reboot my pc to confirm this, I just can't right now because I'm in French class online and they don't like it when you leave lmao)

## System Information:

Option            | Value
------------------|------------------
Operating System  |  Arch Linux x86_64 5.8.12-arch1-1
Software Version  |  v0.4.0-r54.g7fcfefe
Tablet            |  XP-Pen Star G640

X9VoiD, 2020-10-05:
Yes, this is due to a kernel upgrade. `/dev/uinput` will be missing and you'll have to either reboot ore recreate the `dev/uinput`

solonovamax, 2020-10-05:
that's what I thought.
I'm not asking for this to be fixed, just for it to log "restart your computer you complete *buffoon*; you upgraded your kernel"

X9VoiD, 2020-10-05:
Daemon is already logging this, with `Failed to initialize virtual tablet/pointer (ENODEV)`

solonovamax, 2020-10-05:
`otd log` and `otd getcurrentlog` don't show anything and when I was looking through it, I didn't see it being logged.

X9VoiD, 2020-10-05:
that happened because daemon has not restarted yet to see that the `dev/uinput` is already gone

solonovamax, 2020-10-05:
/shrug, that's what I'm proposing to try and log, if possible.

InfinityGhost, 2020-10-06:
I guess we could improve the message to say something for that case, but this is definitely not high priority.

InfinityGhost, 2021-05-25:
I'm fairly certain we've done something about this already, it is at least mentioned in the wiki for sure.

solonovamax, 2021-05-28:
only one way to find out lol

guess I'll just wait for a kernel update.

solonovamax, 2021-05-28:
sick, time to find out.
[image]

solonovamax, 2021-05-28:
idk, I ran otd before updating the system, just to make sure it would work properly (I didn't see why it wouldn't, but best to check), and I think that made it so when starting otd without rebooting, it would work. idk.

gonX, 2022-11-07:
I'm not convinced this is something we can fix smoothly in the program. Since this is [now documented in the wiki](https://opentabletdriver.net/Wiki/FAQ/Linux#fail-virtual-device) I am closing this issue.

closed 2022-11-07 by gonX