OpenTabletDriver/OpenTabletDriver #355: Star G640 Not Detected, even after the fix

KoshakLoL, 2020-10-08:
## Description
My G640 wasn't working, with these errors from the console:

> [DeviceIOException:Error]	Unable to open HID class device (OK).

and

> [ArgumentOutOfRangeException:Error] Value range is [0, 15]. (Parameter 'value')

**What I did:**

1. Followed [this](https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception) guide successfully
2. Rebooted my pc
3. Restarted daemon
4. Ran the tablet debugger, to see if the fix worked...

...It didn't.
I also checked if the tablet and the OTD works in general on Windows 10 - it does.

## System Information:
Option            | Value
------------------|------------------
Operating System  | Manjaro 20.1, 5.8.11
Software Version  | v0.4.1-r13.g9cfa6d9
Tablet            | XP-Pen Star G640

InfinityGhost, 2020-10-11:
We'll have to make sure that the rules are activating as they're supposed to. Did you previously use the XP-Pen drivers?

KoshakLoL, 2020-10-11:
I actually tried to launch the official XP-Pen (V 1.2.13.1) driver after posting this issue and they work fine, but they don't have that many options, so they're not for me. Deleted them after.
But, at the time of posting this, no, I have not tried to install any drivers.

X9VoiD, 2020-10-11:
`rmmod hid-uclogic`
`chmod 0666 /dev/uinput`

then restart daemon and gui. if still not working, join the discord server: https://discord.gg/P3GYdUS

Joehuu, 2020-10-11:
Can confirm the above comment works for me, using Ubuntu 20.04.

InfinityGhost, 2020-10-11:
This appears to be the fix, so I'll be closing this issue. This will be resolved in the next release.

closed 2020-10-11 by InfinityGhost