OpenTabletDriver/OpenTabletDriver #359: Invalid Area in settings.json

X9VoiD, 2020-10-11:
## Description
OTD produces an invalid area configuration on settings.json in cases where there is no tablet to be found, but the "save" button is pressed.

InfinityGhost, 2020-10-11:
What are the side effects of an area being the incorrect size? What could we do for default values instead?

X9VoiD, 2020-10-12:
What happens is that it saves a zero width, zero height rectangle area to the settings.json. What we could do is to block saving a settings.json when the area will result to <NaN, NaN>, or simply when there is no tablet.

InfinityGhost, 2020-10-18:
Unable to reproduce this as of 110cdb181fc895f749d4f780f377c6eeeadcbdeb

InfinityGhost, 2020-11-05:
Reproduced the NaN position by dragging on an area with 0 width & 0 height

commit 5e7ce8ce (2020-11-05): Resolve #359 by prompting with invalid tablet area
closed 2020-11-07 by InfinityGhost