OpenTabletDriver/OpenTabletDriver #363: Failing dotnet build commands under Linux-FAQ

ceilingwaffle, 2020-10-13:
## Description
https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ

./build.sh was failing with error:
[image]
(new setup sorry, unable to copy and paste from terminal)

Updated the wiki here with working commands (not sure how to submit pull requests for wiki pages)
https://github.com/ceilingwaffle/OpenTabletDriver/wiki/Linux-FAQ

```bash
git clone https://github.com/InfinityGhost/OpenTabletDriver.git
git clone https://github.com/InfinityGhost/OpenTabletDriver-udev.git
cd OpenTabletDriver-udev
mkdir ./.modules
mkdir ./build
cp -r ../OpenTabletDriver ./.modules/
./build.sh
sudo mv ./build/30-opentabletdriver.rules /etc/udev/rules.d/30-opentabletdriver.rules
sudo udevadm control --reload-rules
cd ..
rm -rf OpenTabletDriver
rm -rf OpenTabletDriver-udev
```

X9VoiD, 2020-10-13:
Should instead initialize the git submodule.

```git submodule update --init --remote```

InfinityGhost, 2020-10-14:
Updated the wiki to match the submodule update requirement.

closed 2020-10-14 by InfinityGhost