OpenTabletDriver/OpenTabletDriver #364: AutoHotKey

EXtremeExploit, 2020-10-13:
## Description

* Would be nice to instead of use keybinds that you already can do, to be able to do ANY keybind, What i mean is the addition to add AutoHotKey Scripts, When you are about to change a bind it would ask you if you want to setup a keybind or set the path of an AHK file path.

Here is an example i made in paint
[image]
If AHK is selected the keybind input text filed wil get disabled and viceersa

X9VoiD, 2020-10-13:
OTD is expected to work and behave the same on all supported platforms. AutoHotKey is a Windows-only program and so should be implemented as a plugin instead

InfinityGhost, 2020-10-14:
This could be implemented with a IBinding plugin. where you would instead just use the property value to run an AHK script.

skyborgff, 2020-10-14:
@EXtremeExploit can you join the OTD Discord discord.gg/p3gydus and dm me? Skyborg#0894