OpenTabletDriver/OpenTabletDriver #379: does not work in Void Linux

darkralts, 2020-10-16:
## Description

i run the daemon and the gui and the gui cannot connect to the daemon
https://i.imgur.com/qDYAJF1.png
## System Information:

Option            | Value
------------------|------------------
Operating System  | void linux
Software Version  | OpenTabletDriver v0.4.1
Tablet            | 470 wacom

InfinityGhost, 2020-10-17:
That is not v0.4.1 based on that screenshot.

darkralts, 2020-10-19:
i might have ran an older version by accident. will delete both and redownload and try again as i do not know which is which and will repost results.

darkralts, 2020-10-19:
https://i.imgur.com/3WqHyKq.png
this time i got the daemon working but the gui just crashes as soon as i launch it

darkralts, 2020-10-19:
[ralts@pandathighs OpenTabletDriver]$ sudo !!
sudo ./OpenTabletDriver.UX.Gtk
Password: 

(OpenTabletDriver.UX.Gtk:25012): dbind-WARNING **: 01:57:09.304: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-ZoVnpmiQ15: Connection refused
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
System.DllNotFoundException: Unable to load shared library 'libX11' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibX11: cannot open shared object file: No such file or directory
   at OpenTabletDriver.Native.Linux.Xorg.XLib.sys_XOpenDisplay(Char* display)
   at OpenTabletDriver.Native.Linux.Xorg.XLib.XOpenDisplay(Char* display)
   at OpenTabletDriver.Interop.Display.XScreen..ctor()
   at OpenTabletDriver.Interop.Platform.GetLinuxScreen()
   at OpenTabletDriver.Interop.Platform.<>c.<.cctor>b__15_3()
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode mode)
   at System.Lazy`1.ExecutionAndPublication(LazyHelper executionAndPublication, Boolean useDefaultConstructor)
   at System.Lazy`1.CreateValue()
   at OpenTabletDriver.UX.MainForm.ConstructDisplayArea()
   at OpenTabletDriver.UX.MainForm.ConstructMainControls()
   at OpenTabletDriver.UX.MainForm.InitializeAsync()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__139_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Gtk.Application.Run()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.Gtk.MainClass.Main(String[] args)
i got this when i ran with sudo
still crashed but posted this in the daemon terminal this time
[IPC:Debug]	Connected to a client.
[IPC:Debug]	Disconnected from a client.

InfinityGhost, 2020-10-19:
Don't run the GUI or Daemon as root.
```
# assuming the daemon is running
export DISPLAY=:0
./OpenTabletDriver.UX.Gtk
```

InfinityGhost, 2020-10-24:
Closing this, its not an actual issue more so than a cry for help that's now inactive.

closed 2020-10-24 by InfinityGhost

darkralts, 2020-10-24:
true 3; it really is

darkralts, 2020-10-24:
ill let you know if i ever get it to work though

darkralts, 2020-10-24:
[image]
but it also shows this when i try to run it as non root, sorry im not so active

darkralts, 2020-10-24:
nvm im dumb and running the dll,

darkralts, 2020-10-24:
[DeviceIOException:Error]	Unable to open HID class device (OK).
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Driver.<>c__DisplayClass33_0.<FindMatches>b__2(HidDevice device)
   at System.Linq.Utilities.<>c__DisplayClass1_0`1.<CombinePredicates>b__0(TSource x)
   at System.Linq.Utilities.<>c__DisplayClass1_0`1.<CombinePredicates>b__0(TSource x)
   at System.Linq.Utilities.<>c__DisplayClass1_0`1.<CombinePredicates>b__0(TSource x)
   at System.Linq.Enumerable.WhereEnumerableIterator`1.GetCount(Boolean onlyIfCheap)
   at System.Linq.Enumerable.Count[TSource](IEnumerable`1 source)
   at OpenTabletDriver.Driver.TryMatchTablet(TabletProperties tablet, HidDevice& tabletDevice, DeviceIdentifier& identifier, IReportParser`1& parser)
   at OpenTabletDriver.Driver.TryMatch(TabletProperties tablet)
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom PTH-850'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'

heres an error i ran into while running the daemon, then it hung and froze around here, ill try restarting it though

darkralts, 2020-10-24:
yeah it doesnt work, it connects if i use sudo but then when i try to run the gui it will just crash. it doesnt crash with non sudo but the daemon hangs around there so it doesnt rly work.

X9VoiD, 2020-10-24:
Make sure you have a dbus session running.

ssccoonniicc, 2020-10-26:
hey @darkralts can you join the [discord server]( https://discord.gg/P3GYdUS) please so we can help you without the limitations of a message board

darkralts, 2020-10-27:
i will