OpenTabletDriver/OpenTabletDriver #380: More button features

The-Mach34, 2020-10-16:
I am wondering if you could add some more button options and features. For example, I would like to be able to open osu! with one of the buttons on my tablet or do commands like ctrl + shift + t with pen buttons or something.

Mrcubix, 2020-10-16:
basically duplicate of https://github.com/InfinityGhost/OpenTabletDriverPlugins/issues/6

skyborgff, 2020-10-16:
The next release of VoiDPlugins https://github.com/X9VoiD/VoiDPlugins will have the ability to assign a executable / program / file to a button, allowing you to run a AutoHotkey script to run any  command you like

Mrcubix, 2020-10-16:
built today by skyborg https://cdn.discordapp.com/attachments/628651971267788800/766729148898607124/ScriptRunner.dll enjoy

The-Mach34, 2020-10-16:
Cool, thanks.

closed 2020-10-16 by The-Mach34