OpenTabletDriver/OpenTabletDriver #381: DECO 03 Support

Gitkubikon, 2020-10-17:
Hello!!
I have a DECO 03 and since Xp-Pen can't provide good enough drivers, I'd be very graceful if u could support it sooner or later :)
Thanks in advance !!

X9VoiD, 2020-10-17:
Open OTD, then OTD -> Help -> Export Diagnostics

Upload the generated file here

Gitkubikon, 2020-10-17:
[Deco 03 Diagnostics.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5395522/Deco.03.Diagnostics.txt)

X9VoiD, 2020-10-17:
Not sure what's up with your system but the diagnostics failed to retrieve MaxInputReportLength and MaxOutputReportLength, try running daemon as sudo then export diagnostics again

Gitkubikon, 2020-10-17:
Done, I ran it as root

X9VoiD, 2020-10-17:
[Deco 03.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5396130/Deco.03.zip)
Extract zip content, then put it into `Configurations/XP-Pen/Deco 03.json`. Now it might or might not work depending on the initialization needed for the Deco 03, but if it did work, then do the following:

1. Open OTD -> Tablet -> Tablet Debugger
2. Hover your pen to the bottom right most part of the physical tablet, and try to get the maximum value for `Position`
3. Determine the `Report ID` when hovering and not hovering
4. Press your pen until you get max `Pressure`
5. Report back here the `Position`, the max `Pressure`, and the active and inactive `ReportIDs`

X9VoiD, 2020-10-17:
Oh, uh... This configuration format is compatible only on the latest dev build, download it from [here.](https://github.com/InfinityGhost/OpenTabletDriver/actions/runs/309900189)

Gitkubikon, 2020-10-17:
It didn't work, it gave me an error in the console :

DeviceIOException        Unable to open HID class device (OK).

X9VoiD, 2020-10-17:
[Deco 03.zip](https://github.com/InfinityGhost/OpenTabletDriver/files/5396364/Deco.03.zip)

Gitkubikon, 2020-10-17:
[image]

Gitkubikon, 2020-10-17:
[image]

X9VoiD, 2020-10-17:
Join the (discord server)[https://discord.gg/P3GYdUS], DECO 03 might need a different initialization than what's currently known for XP-Pen tablets

InfinityGhost, 2020-10-20:
#387 merged and wasn't linked to this, closing this issue now.

closed 2020-10-20 by InfinityGhost