OpenTabletDriver/OpenTabletDriver #383: The program does not detect the tablet

53jk1, 2020-10-17:
## Description
I installed the package on linux from release .deb. It does not detect any of the tablets for me, I have tried two others. The first is the CTL-480, the second is the CTL-460. The tablet works normally, only the driver does not want to detect the tablet.

## System Information:

Option            | Value
------------------|------------------
System | Ubuntu 20.04.1 LTS (Focal Fossa)
 Software | 0.1.0-rc1
Tablet            |  Wacom CTL-460/CTL-480

X9VoiD, 2020-10-17:
You're using a very old version of OTD, the latest version is v0.4.1

53jk1, 2020-10-17:
It's the same issue at the v0.4.1

console log here
```
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	Searching for tablet 'Huion H430P'
[Detect:Info]	Searching for tablet 'Huion HS610'
[Detect:Info]	Searching for tablet 'Huion H640P'
[Detect:Info]	Searching for tablet 'Huion Inspiroy Q11K'
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion H610 Pro'
[Detect:Info]	Searching for tablet 'Huion HS611'
[Detect:Info]	Searching for tablet 'Huion HS64'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom PTH-451'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom PTH-660'
[Detect:Info]	Searching for tablet 'Wacom PTH-850'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTL-680'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom CTH-470'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom CTL-672'
[Detect:Info]	Searching for tablet 'Wacom PTH-660 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100 Bluetooth'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[DeviceIOException:Error]	Unable to open HID class device (OK).
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[Detect:Info]	Searching for tablet 'Wacom CTL-690'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Wacom CTL-6100'
[Detect:Info]	Searching for tablet 'XP-Pen G640'
[Detect:Info]	Searching for tablet 'XP-Pen G430'
[Detect:Info]	Searching for tablet 'XP-Pen G430S'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]	Searching for tablet 'XP-Pen G960 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]	Searching for tablet 'XP-Pen G640 v2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01 v2'
[Detect:Info]	Searching for tablet 'XP-Pen G640s'
[Detect:Info]	Searching for tablet 'XP-Pen G430S_B'
[Detect:Info]	Searching for tablet 'XP-Pen G540 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen Star 03v2'
```

X9VoiD, 2020-10-17:
https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception

53jk1, 2020-10-17:
Thank you!
Peace <3

closed 2020-10-17 by 53jk1