OpenTabletDriver/OpenTabletDriver #390: Cursor doubling

53jk1, 2020-10-19:
## Description
I installed the entire program, set it up. Everything works, only one little problem that it reads to me like two drivers. Two cursors mapped elsewhere appear, one is as if mapped to two monitors and the other one is mapped only to one. I think it may be the default drivers in Ubuntu.

## System Information:

Option            | Value
------------------|------------------
System | Ubuntu 20.04.1 LTS (Focal Fossa)
 Software | v0.4.1
Tablet            |  Wacom CTL-460/CTL-480

X9VoiD, 2020-10-19:
https://github.com/InfinityGhost/OpenTabletDriver/wiki/Linux-FAQ

53jk1, 2020-10-19:
Well, thanks you!
The first FAQ have solved my problem.
Love ya, peace! :heart:

closed 2020-10-19 by 53jk1