OpenTabletDriver/OpenTabletDriver #394: HidSharp DeviceIOException when tries to open Wacom ISD-V4 on windows 10

Marco-cefetmg, 2020-10-21:
## Description

Error when Hidsharp detects and try to open **Wacom ISD-V4** device.
~~~~
[Detect:Error]	HidSharp.Exceptions.DeviceIOException: Unable to open HID class device (\\?\hid#vid_056a&pid_0093&mi_00&col02#7&1717dee0&2&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}).
   at HidSharp.Platform.Windows.WinHidStream.Init(String path)
   at HidSharp.Platform.Windows.WinHidDevice.OpenDeviceDirectly(OpenConfiguration openConfig)
   at HidSharp.Device.OpenDeviceAndRestrictAccess(OpenConfiguration openConfig)
   at HidSharp.Device.Open(OpenConfiguration openConfig)
   at HidSharp.Device.TryOpen(OpenConfiguration openConfig, DeviceStream& stream, Exception& exception); Retrying 3 more times

~~~~
Complete diagnostics: [diagnostic.json.txt](https://github.com/InfinityGhost/OpenTabletDriver/files/5417423/diagnostic.json.txt)
## System Information:

Option            | Value
------------------|------------------
Operating System  | Windows 10 2004 x64
Software Version  | v0.4.1
Tablet            | Wacom ISD-V4

Mrcubix, 2020-10-22:
we'll need you to join the official discord server, https://discord.gg/P3GYdUS , we will be able to make a proper and probably working config there.

X9VoiD, 2020-10-26:
This issue can be closed now. Wacom ISDV4 of the user cannot be opened in Windows since it is opened exclusively by the OS.

InfinityGhost, 2020-11-28:
Reopening this because we intend to eventually resolve the issue, and it should only be a windows-specific issue.