OpenTabletDriver/OpenTabletDriver #395: TABLET NOT DETECTED PLS HELP :(

supertesticles, 2020-10-22:
Tablet: gaomon s620
idk what's wrong I'm retarded

ssccoonniicc, 2020-10-22:
If you check the console and you see this error: `HidSharp.Exceptions.DeviceIOException: Unable to open HID class device`
Uninstall any other tablet drivers and restart

If no tablets are detected in the console then do this:
Device Manager -> View -> Devices by Connection
Find for any of these devices inside usb controllers:
  -  Gaomon HID
  -  Huion HID
  -  Graphics Tablet Device
  -  HID Tablet
Uninstall them (and check delete driver files) then replug tablet

If you get this error `[Detect:Error]  The configuration directory 'C:\WINDOWS\system32\Configurations' does not exist.`
Find the .exe manually and launch it

supertesticles, 2020-10-23:
> If you check the console and you see this error: `HidSharp.Exceptions.DeviceIOException: Unable to open HID class device`
> Uninstall any other tablet drivers and restart
>
> If no tablets are detected [...]

There was none of the above

ssccoonniicc, 2020-10-23:
Uninstall any other tablet drivers you have and do this:
Device Manager -> View -> Devices by Connection
Find for any of these devices inside usb controllers:

    Gaomon HID
    Huion HID
    Graphics Tablet Device
    HID Tablet
    Uninstall them (and check delete driver files) then replug tablet

supertesticles, 2020-10-23:
Thanks
I couldn't find it because I had to view->Devices by Type and it was under universal bus controllers something like that

closed 2020-10-23 by supertesticles